things4u/ESP-1ch-Gateway #1: Downlink and some other things

rk125, 2019-11-06:
Hello Maarten!

Thank you for long awaited new version!

Couple  things:
- does not compile with "#define STAT_LOG 0"
- would be nice to have on initial display "SSID=xxx IP=x.x.x.x" not "WIFI" instead of "IP", cause that is IP address shown there afterall
- gateway dies immediately after 1st lora packet :
Host esp32-a612cc WiFi Connected to i on IP=192.168.1.160
Local UDP port=1700
Connection successful
Gateway ID: 240AC4FFFFA612CC, Listening at SF9 on 868.10 MHz.
setupOta:: Started
Ready
IP address: 192.168.1.160
Time: Wednesday 19:10:09
Gateway configuration saved
WWW Server started on port 80
OLED_ADDR=0x3C

A readUdp:: NTP msg rcvd
G addLog:: fileno=19, rec=99: 1 4 B4 0 24 A C4 FF FF A6 12 CC {"rxpk":[{"tmst":58315399,"chan":0,"rfch":0,"freq":868.099975,"stat":1,"modu":"LORA","datr":"SF9BW125","codr":"4/5","lsnr":13,"rssi":-90,"size":31,"data":"QIcBIxMAVBAB/Jk20RFKB6FE9iIEh5ajboqhU4ODgQ=="}]}
A readUdp:: Error: PKT_PULL_RESP sendPacket failed
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d3e82  PS      : 0x00060a30  A0      : 0x800db14d  A1      : 0x3ffb1f00
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffc207c  A5      : 0x3ffbecf8
A6      : 0x00000003  A7      : 0x00060423  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000022  A11     : 0x00000000  A12     : 0x3ffc207c  A13     : 0x3ffbecf8
A14     : 0x00000000  A15     : 0x3ffb0060  SAR     : 0x0000000e  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff3

Backtrace: 0x400d3e82:0x3ffb1f00 0x400db14a:0x3ffb1f20 0x400de0cb:0x3ffb1f50 0x400de5ec:0x3ffb1f90 0x400eaf21:0x3ffb1fb0 0x40089341:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

helps changing "JsonObject root = jsonBuffer.to" to "JsonObject root = jsonBuffer.as" as described in v5 issues

- downlink does not work for me. Changing back to v5 with https://github.com/things4u/ESP-1ch-Gateway-v5.0/pull/72 all is ok again
 
All that with Arduino 1.8.9. Gateway TTGO with OLED. ESP32 1.0.4. loraserver.io as server.

Cheers!
Reggie

platenspeler, 2019-11-06:
Hi Reggie,

Thanks for having a first look at version 6.
I will post a new version 6.1.1 probably tomorrow, in that version I will rename ESP-sc-gway.h to configGway.h and sensor.h to configNode.h.
You will find that configNode.h contains all the sensible parts as SSID and password, and 
configGway.h contains all the #define parameters.

I will have a look at your comments below next days.

Thanks,

Maarten

platenspeler, 2019-11-06:
re 1: I do not have that right now. Can you please check with new version tomorrow?
re 2: Done
re 3: Same as 1. I have a TTGO V1 machine here and see no error
re 4: Downlink. The added version ArduinoJSON is Very (very) large. Can you test with the latest version of library manager or tell me what the main difference is between the two versions?

Thanks,

Maarten

rk125, 2019-11-07:
Good evening Maarten!

With 6.1.1.E.EU868; 191106a
1 - same
2 - done, nice!
3 - same:
MAC: 24:0a:c4:a6:12:cc, len=17
WlanConnect:: Init para 0
0:1:4. WiFi connect SSID=i, pass=!
A WlanStatus:: CONNECTED to i
Host esp32-a612cc WiFi Connected to i on IP=192.168.1.160
Local UDP port=1700
Connection successful
Gateway ID: 240AC4FFFFA612CC, Listening at SF9 on 868.10 MHz.
setupOta:: Started
Ready
IP address: 192.168.1.160
Time: Thursday 18:25:53
Gateway configuration saved
WWW Server started on port 80
OLED_ADDR=0x3C

rxPkt:: t=18:25:57, f=0, sf=9, a=38131887, flags=50, addr=0, len=17
R RXDONE in dT=17371000: I=RXDONE HEADER , F=0, SF=9, E=0, S=RX   , eT=17376704, dT=17377833
G addLog:: fileno=0, rec=5: 1 2D CF 0 24 A C4 FF FF A6 12 CC {"rxpk":[{"tmst":17378740,"chan":0,"rfch":0,"freq":868.099975,"stat":1,"modu":"LORA","datr":"SF9BW125","codr":"4/5","lsnr":1,"rssi":-119,"size":17,"data":"QIcYEzgAOAAB7uATVEDdTgI="}]}
A readUdp:: NTP msg rcvd
T sendPacket:: ERROR: data is NULL
A readUdp:: Error: PKT_PULL_RESP sendPacket failed
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d3ec6  PS      : 0x00060e30  A0      : 0x800db441  A1      : 0x3ffb1f00
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffc2090  A5      : 0x3ffbecaf
A6      : 0x00000003  A7      : 0x00060823  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000022  A11     : 0x00000000  A12     : 0x3ffc2090  A13     : 0x3ffbecaf
A14     : 0x00000000  A15     : 0x3ffb0060  SAR     : 0x0000000e  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff3

Backtrace: 0x400d3ec6:0x3ffb1f00 0x400db43e:0x3ffb1f20 0x400de3d3:0x3ffb1f50 0x400de8f4:0x3ffb1f90 0x400eb26d:0x3ffb1fb0 0x40089341:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

4 - same. If I fix "Guru Meditation Error" (which crashes step3) with jsonBuffer.as then gateway works stable, but no downlink. When going back to v5 - ok, downlink working again. I am actually using https://github.com/b00bl1k/ESP-1ch-Gateway-v5.0 branch "fix-downlink" - need to figure out what else except jsonBuffer is changed there.

Normal v5 rx/tx log with working downlink:
rxPkt:: t=18:40:32, f=0, sf=9, a=F9F5B711, flags=50, addr=0, len=19
RXDONE in dT=374899834: I=RXDONE HEADER , F=0, SF=9, E=0, S=RX   , eT=12230475, dT=374902180
G addLog:: fileno=0, rec=25: 1 2F 26 0 24 A C4 FF FF A6 12 CC {"rxpk":[{"tmst":374903171,"chan":0,"rfch":0,"freq":868.099975,"stat":1,"modu":"LORA","datr":"SF9BW125","codr":"4/5","lsnr":13,"rssi":-91,"size":19,"data":"QBG39fkCBwAFAwECRdVcEY0GTw=="}]}
T LoraDown tmst=375903171
M PKT_TX_ACK:: micros=375424373
T TX done:: I=  --  , F=0, SF=9, E=1, S=TX   , eT=938196, dT=375903402
T TXDONE:: rcvd=376088795, diff=185642
T TXDONE:: done OK

Thanks!
Reggie

rk125, 2019-11-07:
Hmm, downlink stopped working on v5 as well, after I enabled debug 2, rx, tx etc. When disabled - all good. Something with timing, I suppose. Need more tests.

Reggie

platenspeler, 2019-11-07:
The machine is VERY sensitive to timing issues. I will check again one of these days to make sure that OLED display and debug settings will nto influende that. Last version I use now has shuffled some of those parts ouside the critical timing parts.
I willl test 1 and 3 shortly to see of they still complie and do not raise exceptions.
Did you more or less use the standard configXXX.h files or changes some parts?

Maarten

rk125, 2019-11-09:
Hello Maarten,

So, "V.6.1.1.E.EU868; 191106a" works for me after all even with downlink, only these things are needed:
- you have nice undocumented feature "Time Correction" in expert mode. When doing debug 2 with rx/tx - downlink worked for me only when I set delay to -15000 (minus). When done with debug - need to set it back to 0 or -5000 for downlink to work. Would be really nice if txDelay would be saved over reboots, for now it resets to 0. All this seems to be affected by serial print on console, not oled.
- for some reason line 163 in _txRx.ino "//LoraDown.fff = freqs[ifreq].dwnFreq;" is commented out. If gw set to strict, frequency is 0 - log shows " Rfreq=868100000, Request=868100000 ->0". I think this is not right and after uncommenting line, downlink stated to work.
- Line 97 in _txRx.ino should say " JsonObject root = jsonBuffer.as<JsonObject>();" otherwise with your's "JsonObject root		= jsonBuffer.to<JsonObject>();" there is meditation error and crash. Might be this is loraserver.io packet which is different, but since others had issue, would be nice if you change it, if it works for you as well.

I needed some changes in config files, just to adjust to my setup. Pinout is 5, Strict set to 1, wifi, timezone and servers.
I'm happy again :)

Reggie

platenspeler, 2019-11-11:
Hi Reggie,

re 1: I am adding to the documentatio. But as you found out it is related to the number of Serial calls in the code. Moving the Serials outside the critical area is unfortunately not always possible.

re 2: Have to look into this, I do not recall why it was taken out. However, testing downling is VERY difficult for me as there are so many good gateways arround.

re 3. I am making the change, but I cannot recall why this should lead to an error.

Maarten

PS. I added another section at the "expert" mode: "LAST SEEN"

closed 2019-12-10 by rk125

platenspeler, 2019-12-26:
Hi Reggie,

Just a question. Should you load the latest version of th egateway, please check the MDNS functions for correctness and the downlink functions (as I have a number of good official gateways in my neighbourhood and I cannot test downlink that easily).

Tx,

Maarten

rk125, 2019-12-26:
Hello Maarten,

Issues:
1. unfortunately seems MDNS is not working without "MDNS.begin(hostname);"
- I added this to ESP-sc-gateway line 546 just after you set
"WiFi.setHostname( hostname );" and totally forgot to mention. Otherwise
all the time MDNS query returns 0.0.0.0. If you want to test yourself -
just run avahi daemon on linux host in local net and set NTP server to
<servername of that host>.local - you will see if it works without breaking
TTN connectivity.
2. does not seem right to "die" in 1st round of for loop starting _WiFi.ino
line 382 - it does not try 2nd and so forth time.
3. line 374 needs "_" in front of MONITOR, right?
4. downlink seems to work at 1st glimpse.

Merry 3rd Xmas!

Best regards,
Reggie

platenspeler, 2019-12-27:
Ooooops,

I removed the die() from the loop and implemented the other changes as well.
All should be OK by now (I hope).

Merry Xmas!

Maarten

platenspeler, 2019-12-27:
Will try your MDNS suggestion later.