things4u/ESP-1ch-Gateway #2: How to add support for IN865-867

bkrajendra, 2019-11-17:
I want to use it for India and need to add 865Mhz support. 
How can I achieve it with this version.

```
IN865-867
Uplink:

865.0625 - SF7BW125 to SF12BW125
865.4025 - SF7BW125 to SF12BW125
865.9850 - SF7BW125 to SF12BW125
Downlink:

Uplink channels 1-3 (RX1)
866.550 - SF10BW125 (RX2)
```

platenspeler, 2019-11-18:
Hi,

Multi-Region support is difficult, especially when there are no sensors available close by.
You can add the frequencies to the loramodem.h file, and if they work please send them to me so that we can adapt the source code and documentation.

What are the frequencies used by India? What are the spreading factors for sending and receiving etc. or are there only three?

regards,

Maarten

PS. I will add something like this to the initial code...

#elif defined(IN865_867)
vector freqs [] = { 
	{ 865062500, 125, 7, 12, 865062500, 125,  7, 12},			// And RX1
	{ 865402500, 125, 7, 12, 865402500, 125,  7, 12},
	{ 865985000, 125, 7, 12, 865985000, 125,  7, 12},
	{         0,   0, 0,  0, 866550000, 125, 10, 10}			// RX2
};

platenspeler, 2019-11-18:
Done

bkrajendra, 2019-11-18:
I'll try this and will provide feedback here.
Thanks a lot!

bkrajendra, 2019-11-18:
More information about Indian LoRa can be found at http://www.ensembletech.in/lora-frequency-bands-india/

[image]

iotlearner0level, 2019-11-19:
hi! is it possible to use 433mhz frequency? Another thing, aliexpress lists uart based modules https://www.aliexpress.com/item/32802241921.html will this library support these? it will save lots of pins and timings if it does making it easier for hobbyists..

platenspeler, 2019-11-25:
I did add EU433 to the server files in "configGway.h". It works, but since 433 is not yet standardized in Europe the frequencies are my definitions for the moment. But you can add your own.

platenspeler, 2019-11-25:
The frequencies used in the link are Loriots and not The Things Network. In fact they seem to differ a lot. I have implemented thos of the Things Network, but you can add you own if you want in the "loraModem.h" file.

platenspeler, 2019-12-16:
Did you get it to work?

bkrajendra, 2019-12-16:
not yet.
Im waiting for my 865Mhz  hardware to arrive.
Will update once I test.

platenspeler, 2020-02-23:
Update?

bkrajendra, 2020-02-24:
currently your latest version is not working with ESP32 and ESP8266.
It compiles but crashes after boot and keeps on restarting. hence unable to test my new hardware with it.

platenspeler, 2020-02-24:
I see in your other message that the module crashes after upload. What loraModem.h settings are your using?
Any major changes to configGway.h and/or configNode.h

Maarten

platenspeler, 2020-02-24:
I uploaded my latest version to both a ESP8266 and a ESP32 gateway with the India settings as in loraModem.h.

I cannot see if the gateway works as I do not have Indian nodes configured.

However, the code does not cause ANY panics on both types of hardware.

bkrajendra, 2020-02-27:
Hi thanks for update.
I have not made nay major changes.
Now as I have proper RFM95 module with SX1276, I uploaded code as it is.
Just changes settings for WiFimnager.
with WiFimnager enabled and disabled in both case i get crash.

platenspeler, 2020-02-29:
So what board do you use? You cannot choose ESP8266 or ESP32 as you like, must suit the hardware. What _PIN_OUT did you select?

Maarten

iotlearner0level, 2020-04-28:
> I did add EU433 to the server files in "configGway.h". It works, but since 433 is not yet standardized in Europe the frequencies are my definitions for the moment. But you can add your own.

hi, just to let you know, i downloaded the latest v (i think 6.4 or something) and eu433 seems to work by just using #define otaa doesn't seem to work for know for me. i'm using euro 868 server because in the ttn, i don't find anyway to use 433mhz. thanks for the update. i use mcci lmic (modified for eu433 manually)

closed 2020-05-16 by platenspeler