things4u/ESP-1ch-Gateway #3: Problem on TTGO T-Beam :: need help

PreinfalkG, 2019-11-18:
Hi things4u

!!! THANKS for your work !!! 
It's a great project and I would like to get it running on my TTGO T-Beam v1.0

i compiled your project and started it on my TTGO T-Beam v1.0
But now I'm stuck in an endloop boot loop -> see 'DEBUG Output' below (ERROR hostByName NTP) ...
furthermore I also have some problems with the hardware configuration (_PIN_OUT==x, OLED, ...). 
that would be my exact hardware https://github.com/lewisxhe/TTGO-T-Beam  -> someone can help me here ...
note: the Paxcounter (https://github.com/cyberman54/ESP32-Paxcounter) runs quite well on this Board!

`
..:: DEBUG Output ::..
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
ESP32 defined, freq=868100000
ARDUINO_ARCH_ESP32 defined
SPIFFS init success
readConfig:: Starting 
.readConfig:: reading line=SSID=ADW20-IoT
SSID=	ADW20-IoT
.readConfig:: reading line=PASS=xxXXxx
PASS=	iotADW20
.readConfig:: reading line=CH=0
CH=	0
.readConfig:: reading line=SF=9
SF=	9
.readConfig:: reading line=FCNT=0
FCNT=	0
.readConfig:: reading line=DEBUG=1
DEBUG=	1
.readConfig:: reading line=PDEBUG=64
PDEBUG=64
.readConfig:: reading line=CAD=1
CAD=1
.readConfig:: reading line=HOP=0
HOP=0
.readConfig:: reading line=NODE=0
NODE=	0
.readConfig:: reading line=BOOTS=202
BOOTS=	202
.readConfig:: reading line=RESETS=0
RESETS=	0
.readConfig:: reading line=WIFIS=203
WIFIS=	203
.readConfig:: reading line=VIEWS=0
VIEWS=	0
.readConfig:: reading line=REFR=0
REFR=	0
.readConfig:: reading line=REENTS=0
REENTS=	0
.readConfig:: reading line=NTPETIM=0
NTPETIM=	0
.readConfig:: reading line=NTPERR=0
NTPERR=	0
.readConfig:: reading line=NTPS=0
NTPS=	0
.readConfig:: reading line=FILEREC=0
FILEREC=	0
.readConfig:: reading line=FILENO=0
FILENO=	0
.readConfig:: reading line=FILENUM=0
FILENUM=	0
.readConfig:: reading line=DELAY=0
DELAY=	0
.readConfig:: reading line=EXPERT=0
EXPERT=	0
readConfig:: Fini

Assert=Do Asserts
debug=1
readConfig:: Starting 
.readConfig:: reading line=SSID=ADW20-IoT
SSID=	ADW20-IoT
.readConfig:: reading line=PASS=xxXXxx
PASS=	iotADW20
.readConfig:: reading line=CH=0
CH=	0
.readConfig:: reading line=SF=9
SF=	9
.readConfig:: reading line=FCNT=0
FCNT=	0
.readConfig:: reading line=DEBUG=1
DEBUG=	1
.readConfig:: reading line=PDEBUG=64
PDEBUG=64
.readConfig:: reading line=CAD=1
CAD=1
.readConfig:: reading line=HOP=0
HOP=0
.readConfig:: reading line=NODE=0
NODE=	0
.readConfig:: reading line=BOOTS=202
BOOTS=	202
.readConfig:: reading line=RESETS=0
RESETS=	0
.readConfig:: reading line=WIFIS=203
WIFIS=	203
.readConfig:: reading line=VIEWS=0
VIEWS=	0
.readConfig:: reading line=REFR=0
REFR=	0
.readConfig:: reading line=REENTS=0
REENTS=	0
.readConfig:: reading line=NTPETIM=0
NTPETIM=	0
.readConfig:: reading line=NTPERR=0
NTPERR=	0
.readConfig:: reading line=NTPS=0
NTPS=	0
.readConfig:: reading line=FILEREC=0
FILEREC=	0
.readConfig:: reading line=FILENO=0
FILENO=	0
.readConfig:: reading line=FILENUM=0
FILENUM=	0
.readConfig:: reading line=DELAY=0
DELAY=	0
.readConfig:: reading line=EXPERT=0
EXPERT=	0
readConfig:: Fini

MAC: 24:6f:28:99:be:28, len=17
WlanConnect:: Init para 0
0:1:3. WiFi connect SSID=ADW20-IoT, pass=xxXXxx
A WlanStatus:: CONNECTED to ADW20-IoT
Host esp32-99be28 WiFi Connected to ADW20-IoT on IP=10.10.10.77
Local UDP port=1700
Connection successful
Gateway ID: 246F28FFFF99BE28, Listening at SF9 on 868.10 MHz.
Setup:: ERROR hostByName NTP
abort() was called at PC 0x400db3ae on core 1

Backtrace: 0x4008cc2c:0x3ffb1ed0 0x4008ce5d:0x3ffb1ef0 0x400db3ae:0x3ffb1f10 0x400ddb59:0x3ffb1f30 0x400e7c83:0x3ffb1fb0 0x40089341:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_d...................
..................
...............
`

platenspeler, 2019-11-22:
So what configGway.h configNode.h are you using. If the pin-out is not correct then the gateway complies but does not work, _PIN_OUT==5 seems most appropriate.

iotlearner0level, 2019-11-23:
ttgo t-beam has a power management chip, which requires some initialisation for different radios, gps, lora, oled individually. Check this: https://github.com/kizniche/ttgo-tbeam-ttn-tracker

platenspeler, 2019-11-25:
Hi,

The T-beam is compiling and working OK as of release 6.1.4. This version works on my version and probably should work on yours as well. Please set "#define GATEWAYNODE 1" in configGway.h
Unfotunately I cannot keep track of all the differences between T-Beam and its pin-out. That's why I ask for pin numbers.

platenspeler, 2019-12-16:
How is the T-Beam doing?

Seeelefant, 2019-12-26:
T-Beam pinout can be found here https://github.com/kizniche/ttgo-tbeam-ttn-tracker/blob/master/main/configuration.h .

closed 2020-11-16 by platenspeler