things4u/ESP-1ch-Gateway #4: Heltech ESP32 error just after the NTP

vangalvin, 2019-11-28:
I hae just tried to set this up on one of my Heltec ESP32 modules. for some reason the first issue is with the NTP server name. I tried the default one and used the NZ NTP server name however neither seemed to work.

The next issue seems to be just after that where the system errors out and reboots.

Any sugestions as to how to track this down?

`A WlanStatus:: CONNECTED to SPARK-XT6HVY
Host esp32-65e438 WiFi Connected to SPARK-XT6HVY on IP=192.168.1.78
Local UDP port=1700
Connection successful
Gateway ID: D8A01DFFFF65E438, Listening at SF9 on 868.10 MHz.
Setup:: ERROR hostByName NTP
abort() was called at PC 0x400dc802 on core 1

Backtrace: 0x4008cc20:0x3ffb1ed0 0x4008ce51:0x3ffb1ef0 0x400dc802:0x3ffb1f10 0x400de5d1:0x3ffb1f30 0x400e9cdf:0x3ffb1fb0 0x40088ba1:0x3ffb1fd0

`

platenspeler, 2019-11-28:
The site "nz.pool.ntp.org" should work in New Zealand (NZ). If not, look at your DNS server setup and whether it is working ok.

vangalvin, 2019-11-29:
Thanks @platenspeler  will take a look, I know it works on other devices so not quite sure why its not working on this one.

vangalvin, 2019-11-29:
mind you the concerning one is the abort() when the system then reboots and throws the back trace. Not sure if this is due to the failure of the NTP so will fix that first.

platenspeler, 2019-11-29:
When the NTP server fails to get a name from the network (WiFi.hostByName(NTP_TIMESERVER, ntpServer line 629 ESP-sc-gway.ino) it calls die().
So put all your debug code to find out whats happening in that section just before die() :-)

platenspeler, 2019-12-16:
Does it work now?

vangalvin, 2019-12-16:
> Does it work now?

I have not had the chance to set it back up yet, silly season is upon us. Hopefully I will get the chance to set it up this weekend and see if that sorts the issue out. 

Fingers crossed it will work :)

platenspeler, 2019-12-17:
Thanks for update.

platenspeler, 2020-01-22:
The latest version has new timing and Wlan code setup. It should work now much better.

platenspeler, 2020-01-29:
I have rewritten parts if the WLAN and NTP parts. They work much faster for me now. Please let me know what you think.

platenspeler, 2020-02-04:
I will close this issue, since I have changed a lot of code since December. I did not encounter this issue on my gateways in January.

closed 2020-02-04 by platenspeler