things4u/ESP-1ch-Gateway #5: mDNS lookup on esp32

rk125, 2019-12-10:
Hello Maarten!

Old problem with mDNS on ESP32. If use "loraserver.local" as ttnserver on ESP8266 - all good, works, but not on ESP32, dies cause cannot resolve hostname. You might not see issue if using public servers, those work always, but mDNS causes problem. On setup where there is no DNS server mDNS is very useful.
Would you consider something like this in the code (better code up to you :), please? 
I added it into _utils for myself. Seems to be working for both esp8266 and esp32 (ttgo) boards I have.

IPAddress resolveHost(char * svr) {
  IPAddress svrIP;
  String svrName = String(svr);
  Serial.print("Server ");
  Serial.println(svrName);
  if (svrName.endsWith(".local")) {
#if ESP32_ARCH==1
    svrName=svrName.substring(0,svrName.length()-6);
    svrIP = MDNS.queryHost(svrName);
    for (byte i=0; i<5; i++) {
      svrIP = MDNS.queryHost(svrName);
      if (svrIP.toString() != "0.0.0.0") break;
      Serial.println("ReTrying to resolve mDNS");
      delay(1000);
    }
#else
    if (!WiFi.hostByName(svr, svrIP))      // Use DNS to get server IP once
    {
      die("Setup:: ERROR hostByName");
    };
#endif
  } else {
    if (!WiFi.hostByName(svr, svrIP))      // Use DNS to get server IP once
    {
      die("Setup:: ERROR hostByName");
    };
  }
  return svrIP;
}

Call it for all 3 servers, ntp, ttn and thingserver, cause for me ntp is also .local on server.
From setup:
  ttnServer=resolveHost(_TTNSERVER);

Thanks!
Reggie

strech345, 2019-12-16:
@rk125 many thanks. Now it works also for me :-)

rk125, 2019-12-16:
Yeee! Glad to hear!
Reggie

platenspeler, 2019-12-16:
I will have a look. I/m busy with 6.1.5 also, so looking whether I can squeeze it in.

Maarten

platenspeler, 2019-12-16:
Added the code to 6,1,5 together with some clight mods

Geert-69, 2019-12-21:
Can someone tell what files to modify and add lines where ?
Thanks

closed 2019-12-27 by platenspeler