things4u/ESP-1ch-Gateway #6: board reboot from API

rk125, 2019-12-10:
Maarten,

I experience issue that at some random time gateway kinda works, sends stats to loraserver, but no lora data received. Power cycle helps.
I added:

  // reboot
  server.on("/REBOOT", []() {
    sendWebPage("","");             // Send the webPage string
    server.sendHeader("Location", String("/"), true);
    server.send ( 302, "text/plain", "");
    ESP.restart();
  });

to _wwwServer.
Then I can monitor packets received on node-red for example and send reboot to gateway which fixes issue. Not perfect, but works.
At least I did not see any other way to reboot gateway board. Or is there any?
Would you consider adding this to code, please?

Thanks!
Reggie

platenspeler, 2019-12-15:
I take it that you want to included a REBOOT option in the Gateway Settings as well? We need a button to start the Reboot sequence.

rk125, 2019-12-16:
Hello Maarten,
For my need enough that we have this via web call. For general use button could help as well for sure.
Thank you!
Reggie

platenspeler, 2019-12-16:
Done in 6.1.5

closed 2019-12-16 by platenspeler