things4u/ESP-1ch-Gateway #7: Error with ESP32 and WIFIMANAGER=1

Alpengroove, 2019-12-17:
Hi!
Wenn compiling the V6.0 sketch on a TTGO ESP32 Lora board with:

#define _WIFIMANAGER 1

I get the following error, when WIFIMANAGER=0, the sketch can be compiled without errors (Tnx for te sketch!):

ESP-1ch-Gateway-v6.0/libraries/WiFiManager/WiFiManager.h:22:30: fatal error: user_interface.h: No such file or directory

I guess the lib is only for ESP8266? Is there a solution for an ESP32?

Seeelefant, 2019-12-26:
Same here, could https://github.com/Links2004/arduinoWebSockets/issues/205 solve this problem?

platenspeler, 2019-12-30:
Hi,

I will have a look. As you mention, only ESP8266 is supported at the moment.
But there seem to be more implementations of wifimanager.

Maarten

Ps. In any case, please download wifimanager from Tzapu from the on-line ESP8266 library.

platenspeler, 2020-01-29:
See my last code version 6.2.0 which should work better.

platenspeler, 2020-01-29:
But still, for ESP32 I might have to change parts of the source code.

knopserl, 2020-02-17:
@all I have/had the same problem. If you define _WIFIMANAGER  1 then there is a collition between ESP_WiFiManager and the ESP32WebServer library. duplicate definitions etc.
So I replaced the ESP_WiFiManager by the standard WiFiManager (https://github.com/tzapu/WiFiManager) and also replaced the ESP32WebServer with the ESP32 default WebServer.h from the ESP32 package which has very recent maintenance in comparison to the ESP32WebServer.

So my ESP-sc-gway looks like this: 
````
// ----------- Specific ESP32 stuff --------------
#if defined (ARDUINO_ARCH_ESP32) || defined(ESP32)
#	define ESP32_ARCH 1
# include <WiFi.h>
#	include <ESPmDNS.h>
#	include <SPIFFS.h>

#	if _WIFIMANAGER==1
#		define ESP_getChipId()   ((uint32_t)ESP.getEfuseMac())
//#		include <ESP_WiFiManager.h>							// Library for ESP WiFi config through an AP
#   include <WiFiManager.h>  //https://github.com/tzapu/WiFiManager
#	endif //_WIFIMANAGER

#	if A_SERVER==1
#   include <WebServer.h>
#		include <Streaming.h>          						// http://arduiniana.org/libraries/streaming/
		WebServer server(A_SERVERPORT);
#	endif //A_SERVER
````

and in the _WiFI the wifiMgr looks like that:

````
// ----------------------------------------------------------------------------
// When ESP WiFi Manager, do this
// ----------------------------------------------------------------------------
int wifiMgr() 
{

#if _WIFIMANAGER==1
	WiFiManager wifiManager;
	
#	if _MONITOR>=1
	if (debug>=1) {
		mPrint("Starting Access Point Mode");
		mPrint("Connect Wifi to accesspoint and connect to IP: 192.168.4.1");
	}
#	endif //_MONITOR

  wifiManager.setDebugOutput(false);
  wifiManager.setAPCallback(configModeCallback);

  //and goes into a blocking loop awaiting configuration
  if (!wifiManager.autoConnect()) {
    Serial.println("failed to connect and hit timeout");
    ESP.restart();
    delay(1000);
  }
````

and then all works perfect! Thanks to @platenspeler for this great project!

platenspeler, 2020-02-23:
Thanks. I have made some changes to the source, However, the code is still in development so it is not possble to get a stable version. However we might expect that the current functionality will stay,

A lot is possible at this moment, but modifying the SSID through wifimanager for ESP32 for some reason seems to lockup the user interface, So that still needs more work.