things4u/ESP-1ch-Gateway #8: Downlink not working

orinizri1, 2019-12-25:
Dear everyone,
I installed the gateway on TTGO LoRa32 V1 moudle and everything worked fine except of the downlink. I successfully send uplinks to the TTN, but cant send downlink via TTN.
When i configured the gateway CAD was 1 and STRICT_1CH  is 1 too. Wifi Manager is 0.
Im simulating downlink via TTN, i see the scheduled message on the console, but the downlink doesn't sent. I keep seeing the uplinks arrived. I set the gateway to class A.
Any tips? Any way to debug it?
Ori

RobertSasak, 2020-02-21:
My observations are. I can see JOIN followed by downlink JOIN ACCEPT in TTN traffic tab. I can also see couple of packages in **Packages Downlink**. There are no other gateways near by. My weather station does not deliver it status to TTN. From what I read online it could be because it cannot join TTN. And this can be because JOIN ACCEPT is not delivered via downlink. 

I can see two options. Try to change some config paramters in header files. 
I would like to try v5 first. https://github.com/things4u/ESP-1ch-Gateway-v5.0
This issue is closed but describe downlink problems https://github.com/things4u/ESP-1ch-Gateway/issues/1

halukmy, 2020-02-24:
downlink not fukin works

platenspeler, 2020-04-27:
I have included version 6.2.4 which works with downlink. Please test and tell me if I need to update.

closed 2020-04-27 by platenspeler

halukmy, 2020-04-27:
@platenspeler  have you got chirpstack support with CLASS C? class c is so important bro