things4u/ESP-1ch-Gateway #9: resolveHost:: ERROR hostByName

Seeelefant, 2019-12-26:
Trying to run the gateway on a TTGO T-Beam (even with adjusted pins) I get resolveHost:: ERROR hostByName .

11:24:47.509 -> readConfig:: Starting 
11:24:47.544 -> .SSID=	XXXXXXX
11:24:47.544 -> .PASS=	YYYYYYY
11:24:47.544 -> .CH=	0
11:24:47.544 -> .SF=	9
11:24:47.544 -> .FCNT=	0
11:24:47.544 -> .DEBUG=	1
11:24:47.544 -> .PDEBUG=	64
11:24:47.544 -> .CAD=	1
11:24:47.544 -> .HOP=	0
11:24:47.544 -> .NODE=	0
11:24:47.544 -> .BOOTS=	1
11:24:47.544 -> .RESETS=	0
11:24:47.544 -> .WIFIS=	3
11:24:47.544 -> .VIEWS=	0
11:24:47.544 -> .REFR=	0
11:24:47.544 -> .REENTS=	0
11:24:47.544 -> .NTPETIM=	0
11:24:47.544 -> .NTPERR=	0
11:24:47.544 -> .NTPS=	0
11:24:47.544 -> .FILEREC=	0
11:24:47.544 -> .FILENO=	0
11:24:47.544 -> .FILENUM=	0
11:24:47.544 -> .DELAY=	0
11:24:47.544 -> .TRUSTED=	1
11:24:47.544 -> .EXPERT=	0
11:24:47.544 -> .SEEN=	0
11:24:47.544 -> .MONITOR=	0
11:24:47.544 -> readConfig:: Fini
11:24:47.544 -> 
11:24:47.544 -> MAC: 24:6f:28:97:91:e0, len=17
11:24:47.654 -> 0:1:3. WiFi connect SSID=XXXXXXX, pass=YYYYYYY
11:24:56.683 -> Host esp32-9791e0 WiFi Connected to XXXXXXX on IP=192.168.0.38
11:24:57.867 -> Gateway ID: 246F28FFFF9791E0, Listening at SF9 on 868.10 MHz.
11:24:58.155 -> resolveHost:: ERROR hostByName
11:24:58.228 -> abort() was called at PC 0x400dddf6 on core 1
11:24:58.228 -> 
11:24:58.228 -> Backtrace: 0x4008f200:0x3ffb1e30 0x4008f431:0x3ffb1e50 0x400dddf6:0x3ffb1e70 0x400ddf5e:0x3ffb1e90 0x400e1706:0x3ffb1f30 0x400ef47f:0x3ffb1fb0 0x4008b915:0x3ffb1fd0
11:24:58.228 -> 
11:24:58.228 -> Rebooting...
11:24:58.261 -> ets Jun  8 2016 00:22:57
11:24:58.261 -> 
11:24:58.261 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
11:24:58.261 -> configsip: 0, SPIWP:0xee
11:24:58.261 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
11:24:58.261 -> mode:DIO, clock div:1
11:24:58.261 -> load:0x3fff0018,len:4
11:24:58.261 -> load:0x3fff001c,len:1044
11:24:58.261 -> load:0x40078000,len:8896
11:24:58.261 -> load:0x40080400,len:5816
11:24:58.261 -> entry 0x400806ac
11:25:01.628 -> readConfig:: Starting 
11:25:01.664 -> .SSID=	XXXXXXX
11:25:01.699 -> .PASS=	YYYYYYY
11:25:01.699 -> .CH=	0
11:25:01.699 -> .SF=	9
11:25:01.699 -> .FCNT=	0
11:25:01.699 -> .DEBUG=	1
11:25:01.699 -> .PDEBUG=	64
11:25:01.699 -> .CAD=	1
11:25:01.699 -> .HOP=	0
11:25:01.699 -> .NODE=	0
11:25:01.699 -> .BOOTS=	2
11:25:01.699 -> .RESETS=	0
11:25:01.699 -> .WIFIS=	4
11:25:01.699 -> .VIEWS=	0
11:25:01.699 -> .REFR=	0
11:25:01.699 -> .REENTS=	0
11:25:01.699 -> .NTPETIM=	0
11:25:01.699 -> .NTPERR=	0
11:25:01.699 -> .NTPS=	0
11:25:01.699 -> .FILEREC=	0
11:25:01.699 -> .FILENO=	0
11:25:01.699 -> .FILENUM=	0
11:25:01.699 -> .DELAY=	0
11:25:01.699 -> .TRUSTED=	1
11:25:01.699 -> .EXPERT=	0
11:25:01.699 -> .SEEN=	0
11:25:01.699 -> .MONITOR=	0
11:25:01.699 -> readConfig:: Fini
11:25:01.699 -> 
11:25:01.769 -> WARNING:: readSeen, history file not exists /gwayNum.txt
11:25:02.830 -> readConfig:: Starting 
11:25:02.865 -> .SSID=	XXXXXXX
11:25:02.865 -> .PASS=	YYYYYYY
11:25:02.865 -> .CH=	0
11:25:02.865 -> .SF=	9
11:25:02.865 -> .FCNT=	0
11:25:02.865 -> .DEBUG=	1
11:25:02.865 -> .PDEBUG=	64
11:25:02.865 -> .CAD=	1
11:25:02.865 -> .HOP=	0
11:25:02.865 -> .NODE=	0
11:25:02.865 -> .BOOTS=	2
11:25:02.865 -> .RESETS=	0
11:25:02.865 -> .WIFIS=	4
11:25:02.865 -> .VIEWS=	0
11:25:02.865 -> .REFR=	0
11:25:02.865 -> .REENTS=	0
11:25:02.865 -> .NTPETIM=	0
11:25:02.865 -> .NTPERR=	0
11:25:02.865 -> .NTPS=	0
11:25:02.865 -> .FILEREC=	0
11:25:02.865 -> .FILENO=	0
11:25:02.865 -> .FILENUM=	0
11:25:02.865 -> .DELAY=	0
11:25:02.865 -> .TRUSTED=	1
11:25:02.865 -> .EXPERT=	0
11:25:02.865 -> .SEEN=	0
11:25:02.865 -> .MONITOR=	0
11:25:02.865 -> readConfig:: Fini
11:25:02.865 -> 
11:25:02.865 -> MAC: 24:6f:28:97:91:e0, len=17
11:25:02.975 -> 0:1:3. WiFi connect SSID=XXXXXXX, pass=YYYYYYY

Geert-69, 2019-12-26:
The same here:

resolveHost:: ERROR hostByName

platenspeler, 2019-12-27:
Oops,

I found two nasty typos in _sensors.ino for local sensors ("define GATEWAYNODE 1"), but that is not your problem apparently.
The code compiles and runs OK for my T-beam, but it does not for yours. So I added some text in the code, please run again.

Did it run OK in earlier versions?

Maarten

Seeelefant, 2019-12-27:
Hi Maarten,
thank you, looks much better now:

14:18:05.931 -> readConfig:: Starting 
14:18:05.931 -> .SSID=	YYYYYYY
14:18:05.963 -> .PASS=	XXXXX
14:18:05.963 -> .CH=	0
14:18:05.963 -> .SF=	9
14:18:05.963 -> .FCNT=	0
14:18:05.963 -> .DEBUG=	1
14:18:05.963 -> .PDEBUG=	64
14:18:05.963 -> .CAD=	1
14:18:05.963 -> .HOP=	0
14:18:05.963 -> .NODE=	0
14:18:05.963 -> .BOOTS=	185
14:18:05.963 -> .RESETS=	0
14:18:05.963 -> .WIFIS=	189
14:18:05.963 -> .VIEWS=	0
14:18:05.963 -> .REFR=	0
14:18:05.963 -> .REENTS=	0
14:18:05.963 -> .NTPETIM=	17
14:18:05.963 -> .NTPERR=	3
14:18:05.963 -> .NTPS=	9
14:18:05.963 -> .FILEREC=	0
14:18:05.963 -> .FILENO=	0
14:18:05.963 -> .FILENUM=	0
14:18:05.963 -> .DELAY=	0
14:18:05.963 -> .TRUSTED=	1
14:18:05.963 -> .EXPERT=	0
14:18:05.963 -> .SEEN=	0
14:18:05.963 -> .MONITOR=	0
14:18:05.963 -> readConfig:: Fini
14:18:05.963 -> 
14:18:06.980 -> readConfig:: Starting 
14:18:07.046 -> .SSID=	YYYYYYY
14:18:07.080 -> .PASS=	XXXXX
14:18:07.080 -> .CH=	0
14:18:07.080 -> .SF=	9
14:18:07.080 -> .FCNT=	0
14:18:07.080 -> .DEBUG=	1
14:18:07.080 -> .PDEBUG=	64
14:18:07.080 -> .CAD=	1
14:18:07.080 -> .HOP=	0
14:18:07.080 -> .NODE=	0
14:18:07.080 -> .BOOTS=	185
14:18:07.080 -> .RESETS=	0
14:18:07.080 -> .WIFIS=	189
14:18:07.080 -> .VIEWS=	0
14:18:07.080 -> .REFR=	0
14:18:07.080 -> .REENTS=	0
14:18:07.080 -> .NTPETIM=	17
14:18:07.080 -> .NTPERR=	3
14:18:07.080 -> .NTPS=	9
14:18:07.080 -> .FILEREC=	0
14:18:07.080 -> .FILENO=	0
14:18:07.080 -> .FILENUM=	0
14:18:07.080 -> .DELAY=	0
14:18:07.080 -> .TRUSTED=	1
14:18:07.080 -> .EXPERT=	0
14:18:07.080 -> .SEEN=	0
14:18:07.080 -> .MONITOR=	0
14:18:07.080 -> readConfig:: Fini
14:18:07.080 -> 
14:18:07.080 -> MAC: 24:6f:28:97:91:e0, len=17
14:18:07.182 -> 0:1:3. WiFi connect SSID=YYYYYYY, pass=XXXXX
14:18:16.232 -> Host esp32-9791e0 WiFi Connected to YYYYYYY on IP=192.168.0.38
14:18:17.425 -> Gateway ID: 246F28FFFF9791E0, Listening at SF9 on 868.10 MHz.
14:18:58.670 -> writeGwayCfg: Friday 14:18:58
14:18:58.704 -> WWW Server started on port 80
14:19:02.940 -> OLED_ADDR=0x3C
14:19:02.940 -> --------------------------------------

Whats currently not working is:
WIFIMANAGER=1
trying to compile I get 
 Used: C:\Users\xx\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
C:\Users\xx\Documents\Arduino\libraries\ESP8266WiFi\src/ESP8266WiFiType.h:26:19: fatal error: queue.h: No such file or directory

 Not used: C:\Users\xx\Documents\Arduino\libraries\ESP8266WiFi
compilation terminated.

OLED is currently not working, though I adapted 
#elif _PIN_OUT==5								// TTGO (onboard version used, also for DIY)
#define OLED_SCL 22								// GPIO15 / 
#define OLED_SDA 21								// GPIO4 / 
#define OLED_RST 16								// Reset pin (Some OLED displays do not have it)

With "adjusted Pins" I meant: I have a more recent T-Beam and I tried to adapt the pins according to 
https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/Platform_ESP32.h

Seeelefant, 2019-12-27:
I am also not sure if GPS  is working, with the code in this repository https://github.com/lyusupov/SoftRF  it was working.

Geert-69, 2019-12-27:
On Heltec LoRa 32 (v2)

resolveHost:: ERROR hostByName=your.extra.server

Geert-69, 2019-12-27:
It's working now with Version 6.1..5 i; Remove optional second server

platenspeler, 2019-12-27:
Please read the documentation. The second server is optional, and your.extra.server of in configNode.h as a placeholder. It is NOT a real name.

So if you do not have this server,  do not define it at all ....

Therefore in my latest version of the gateway I did not define it to avoid confusion.

Maarten

Maarten

Seeelefant, 2019-12-28:
AXP code can be found here https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/Platform_ESP32.cpp

platenspeler, 2019-12-28:
Hi,

I will not look into other implementations that much.
The T-Beam works for me, but I have put a wire across pin 33 and pin 22(SCL) in order to get DIO1 connected. How did you do that?

Next you have to figure out how to get OLED working.

Maarten

Seeelefant, 2019-12-28:
Hi Maarten,
no, haven´t done that. For [SoftRF](https://github.com/lyusupov/SoftRF) it worked out of the box and according to https://www.bjoerns-techblog.de/2018/10/ttgo-t-beam-kompakter-knochen-zum-mappen/ it should be no longer necessary. But I can give it a try.
Wolfgang

platenspeler, 2019-12-28:
I read the picture wrong I guess. But pin 33 should be connected to Lora-IO2 as far as I can see.  You have to have DIO1 connected in order to get CAD working.

That means that SDA and SCL should be free for I2C devices like OLED. I cannot see what's wrong with OLED as I do not have that device connected. But with the SDA and SCL devices connected it should work

Seeelefant, 2019-12-29:
Hi Maarten,
finally...LED is working, the secret is the initialization of the AXP192:
```
#include <axp20x.h>

AXP20X_Class pmu;
void AXP192_power(bool on) {
  if (on) {
    pmu.setPowerOutPut(AXP192_LDO2, AXP202_ON);  // Lora on T-Beam V1.0
    pmu.setPowerOutPut(AXP192_LDO3, AXP202_ON);  // Gps on T-Beam V1.0
    pmu.setPowerOutPut(AXP192_DCDC1, AXP202_ON); // OLED on T-Beam v1.0
    // pmu.setChgLEDMode(AXP20X_LED_LOW_LEVEL);
    pmu.setChgLEDMode(AXP20X_LED_BLINK_1HZ);
  } else {
    pmu.setChgLEDMode(AXP20X_LED_OFF);
    pmu.setPowerOutPut(AXP192_DCDC1, AXP202_OFF);
    pmu.setPowerOutPut(AXP192_LDO3, AXP202_OFF);
    pmu.setPowerOutPut(AXP192_LDO2, AXP202_OFF);
  }
}
void AXP192_init() {      
    Wire.begin(21, 22);
    if (pmu.begin(Wire, AXP192_SLAVE_ADDRESS))  {
      Serial.println("AXP192 PMU initialization failed");
    }
    else {
Serial.println("Initialzation...");
      // configure AXP192
      pmu.setDCDC1Voltage(3500);              // for external OLED display
      pmu.setTimeOutShutdown(false);          // no automatic shutdown
      pmu.setTSmode(AXP_TS_PIN_MODE_DISABLE); // TS pin mode off to save power

      // switch ADCs on
      pmu.adc1Enable(AXP202_BATT_VOL_ADC1, true);
      pmu.adc1Enable(AXP202_BATT_CUR_ADC1, true);
      pmu.adc1Enable(AXP202_VBUS_VOL_ADC1, true);
      pmu.adc1Enable(AXP202_VBUS_CUR_ADC1, true);

      // switch power rails on
      AXP192_power(true);

      Serial.println("AXP192 PMU initialized");
    }
}
```
followed by
```
AXP192_init();
AXP192_power(true);
```

in the init section.

Seeelefant, 2019-12-29:
Hi Maarten, just FYI:

Pins | T_BEAM(AXP192) | T_BEAM_V07
-- | -- | --
LORA_SCK | 5 | 5
LORA_MISO | 19 | 19
LORA_MOSI | 27 | 27
LORA_SS | 18 | 18
LORA_DI0 | 26 | 26
LORA_RST | 23 | 23
GPS_RX_PIN | 34 | 12
GPS_TX_PIN | 12 | 15
I2C_SDA | 21 | 21
I2C_SCL | 22 | 22
PMU_IRQ | 35 | N/A
USER BUTTON | 38 | 39
GPS_PPS | 37 | N/A

Modules | T_BEAM_V10 | T_BEAM_V07
-- | -- | --
GPS | LDO3 | No support
LORA | LDO2 | No support
OLED | DCDC1 | No support

Best
Wolfgang

platenspeler, 2020-01-29:
Assuming you still have problems with resolveHost. What are the hostnames you are trying to resolve? See configGway.h file and configNode.h. The _THINGSERVER should be undefined if you are not using a second server.

#define NTP_TIMESERVER "nl.pool.ntp.org"	// Country and region specific
#define _TTNSERVER "router.eu.thethings.network"
// #define _THINGSERVER "your 2nd server"

I did rewrite significant parts of the code so the standard code should work beter now.

Seeelefant, 2020-01-30:
Thank you Maarten, I will give it a try this weekend.
Best
--Wolfgang

platenspeler, 2020-02-20:
Status update?

Seeelefant, 2020-02-20:
Hi Maarten,
now I am able to connect
[image]

What´s currently missing in the current code is the initialization via the AXP192 module and the compile problem for the WIFIMANAGER.
Shall I separate this from the message?

Best
Wolfgang

platenspeler, 2020-02-23:
Please separate these issues!
Thanks for the feedback.

Maarten

closed 2020-02-23 by platenspeler