things4u/ESP-1ch-Gateway #10: compile error

dirk99, 2019-12-30:
hi folks,
everytime when i compile i get an error with missing function checkMIC (undefined reference. I hav tryed this with VSC and arduino ide. how can i solve this?

i fond the function in _sensor but the error cames from _txRx

thanks

platenspeler, 2020-01-29:
What is the exact message from the compiler? Did you include the lcode.h and AES include files and libraries?

btw checkMic is the right spelling..

Maarten

platenspeler, 2020-02-03:
Hi Dirk, 

Is this issue solved? Please provide feedback on my question if not. If this issue is not followed up I need to close the issue.

Maarten

dirk99, 2020-02-04:
Hi Maarten, yes the Problem is solved. Thanks

Mit freundlichen Grüßen
Dirk Howold
Tel.  +491702309712
Mail howold@gmx.de

closed 2020-02-04 by platenspeler