things4u/ESP-1ch-Gateway #11: where to put the Gateway Key ?

sprokkie, 2019-12-31:
got it running, but TTN stil says it is offline.
ill gues i have to put the Gateway Key somewhere to identify my gateway
where do i put that ?

sprokkie, 2019-12-31:
ah found it,
its generated by the software

closed 2019-12-31 by sprokkie

RobertSasak, 2020-02-21:
Do you mind to desribe in more detail how can I register my gateway on TTN page?
I have created gateway on TTN page. There is a long **Gateway Key** on TTN page for my gateway. I would expect that I need to put this key into my Arduino code for my gateway and then compile and upload it. However I do know where.

RobertSasak, 2020-02-21:
I fount it. When creating new Gateway on TTN one have to select checkbox "I'm using the legacy packet forwarder". ID field change and expect Gateway ID. Gateway ID is hex string of length 16 chars. It looks like a MAC address. You can find Gateway ID in the web interface of your gateway. You may need to click button "Expert mode" first.