things4u/ESP-1ch-Gateway #12: Why not Platformio?

arrowcircle, 2020-01-12:
Hi! Arduino IDE is really outdated environment. One of it's problems is dependency management - you need to copy them manually. It's not possible to reliably update and set versions.

Platformio does have library management and env setup is only click compile. It will download all necessary and needed versions of libs and arduino framework.

Also, with platformio its easy to integrate CI service (like Travis CI or Github Actions) to automate code checks (linters, build checks and so on) as well as automatic releases.

For example, marlin 2.0 firmware migrated to platformio, so its not something "special".

Is there are any plans to migrate the development to platformio?

Gehilfe, 2020-01-22:
PlatformIO would be great! I've been trying to get the project running on PlatformIO for about a week, but I'm unfortunately a beginner in this area. Has anyone had success with it?

platenspeler, 2020-02-03:
Version 6.2.1 is output of PlatformIO.

closed 2020-02-20 by platenspeler