things4u/ESP-1ch-Gateway #13: permanent setting to listen on specific channel

webgps, 2020-01-14:
Hi All,

Where can I set in the code, that the gateway listens on specific chanel instead on channel 0. I managed to set on the WEB GUI but since i have SPIFFS format 1 on every boot i have to change the channel.

Tried to set #define EU863_870 1 to #define EU863_870 2 no effect, the gw reports it's still listening on 868.100MHz

Tried to change the first value in freq plan  

> { 868100000, 125, 7, 12, 868300000, 125, 7, 12},			// Channel 0, 868.1 MHz/125 primary
> > 

no effect neither,

Thanks in advance.

webgps, 2020-01-14:
> Hi All,
>
> Where can I set in the code, that the gateway listens on specific chanel instead on channel 0. I managed to set on the WEB GUI but since i have SPIFFS format 1 on every boot i have to [...]

Okay partially my fault, forget to change the first value to 868300000, now after compile an upload the gatway starts to listen on chanel 1 by deault.
But is there any other way beyond the GUI to change this?

platenspeler, 2020-02-20:
Unfortunately, channel 0 is considered standard for gateways. If you like to have other frequency for channel 0, you have to change the .h file. 
I do not think that changing the channel in the configuration file will help much, and I plan to leave it like this.

Maarten.

closed 2020-05-16 by platenspeler