things4u/ESP-1ch-Gateway #16: TTGO ESP32 V2.1 - compile fail

toninog, 2020-01-28:
Hi

Thanks for a great project.  I am trying to get a single channel gateway running on a TTGo LoRa32 v2.1 hardware.  When compiling I get this error:

WARNING: library ESP32WebServer claims to run on ESP32 architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
In file included from /Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/ESP-sc-gway.ino:74:0:
/Users/study/Documents/Arduino/libraries/ESP32WebServer/src/ESP32WebServer.h:162:15: error: cannot declare field 'ESP32WebServer::_server' to be of abstract type 'WiFiServer'
   WiFiServer  _server;
               ^
In file included from /Applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFi.h:32:0,
                 from /Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/ESP-sc-gway.ino:69:
/Applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiServer.h:31:7: note:   because the following virtual functions are pure within 'WiFiServer':
 class WiFiServer : public Server {
       ^
In file included from /Users/study/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Arduino.h:152:0,
                 from /var/folders/_q/8cmyshtd06gfj04gz2t8mp5w0000gn/T/arduino-sketch-A301C915104278530327603DD625734A/sketch/ESP-sc-gway.ino.cpp:1:
/Users/study/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Server.h:28:18: note:    virtual void Server::begin(uint16_t)
     virtual void begin(uint16_t port=0) =0;
                  ^
/Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/ESP-sc-gway.ino: In function 'void setup()':
/Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/ESP-sc-gway.ino:357:7: error: 'class WiFiClass' has no member named 'mode'
  WiFi.mode(WIFI_STA);         // WiFi settings for connections
       ^
/Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/ESP-sc-gway.ino:357:12: error: 'WIFI_STA' was not declared in this scope
  WiFi.mode(WIFI_STA);         // WiFi settings for connections
            ^
/Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/ESP-sc-gway.ino:358:7: error: 'class WiFiClass' has no member named 'setAutoConnect'
  WiFi.setAutoConnect(true);
       ^
/Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/ESP-sc-gway.ino:388:8: error: 'class WiFiClass' has no member named 'setHostname'
   WiFi.setHostname( hostname );
        ^
/Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/ESP-sc-gway.ino:400:28: error: 'class WiFiClass' has no member named 'getHostname'
    response += String(WiFi.getHostname());
                            ^
/Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/_WiFi.ino: In function 'int WlanStatus()':
/Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/_WiFi.ino:40:9: error: 'class WiFiClass' has no member named 'setAutoReconnect'
    WiFi.setAutoReconnect(true);    // Reconect to this AP if DISCONNECTED
         ^
/Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/_WiFi.ino: In function 'int WlanConnect(int)':
/Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/_WiFi.ino:238:9: error: 'class WiFiClass' has no member named 'mode'
    WiFi.mode(WIFI_STA);
         ^
/Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/_WiFi.ino:238:14: error: 'WIFI_STA' was not declared in this scope
    WiFi.mode(WIFI_STA);
              ^
/Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/_WiFi.ino:268:9: error: 'class WiFiClass' has no member named 'persistent'
    WiFi.persistent(false);
         ^
/Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/_WiFi.ino:269:9: error: 'class WiFiClass' has no member named 'mode'
    WiFi.mode(WIFI_OFF);   // this is a temporary line, to be removed after SDK update to 1.5.4
         ^
/Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/_WiFi.ino:269:14: error: 'WIFI_OFF' was not declared in this scope
    WiFi.mode(WIFI_OFF);   // this is a temporary line, to be removed after SDK update to 1.5.4
              ^
/Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/_udpSemtech.ino: In function 'int sendUdp(IPAddress, int, uint8_t*, int)':
/Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/_udpSemtech.ino:362:57: error: invalid operands of types 'const char [35]' and 'char*' to binary 'operator+'
   mPrint("sendUdp: WlanConnect connected to="+WiFi.SSID()+". Server IP="+ String(WiFi.localIP().toString()) );
                                                         ^
/Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/_wwwServer.ino: In function 'void wifiConfig()':
/Users/study/Downloads/development/sensor/lora32/ESP-1ch-Gateway-v6.0/ESP-sc-gway/_wwwServer.ino:1093:19: error: 'class WiFiClass' has no member named 'getHostname'
   response +=WiFi.getHostname(); response+="</tr>";
                   ^
Multiple libraries were found for "ESP32WebServer.h"
 Used: /Users/study/Documents/Arduino/libraries/ESP32WebServer
 Not used: /Users/study/Documents/Arduino/libraries/src
Error during build: exit status 1

Any suggestions on resolving this?

I've tried different versions of the ESP32WebServer library and no success.

I compile using arduino-cli (although the Arduino IDE has the same error)

`arduino-cli compile --fqbn esp32:esp32:ttgo-lora32-v1 ESP-sc-gway`

Thanks

platenspeler, 2020-02-26:
Please use the latest release of the gateway and try again. Please remember that PlatformIO for example seems tio find a lot or "errors" that it does not care about and that are not under my control. 

Anyway, I moved to another version of the webserver and did not anymore include a separate ESPWebserver file.

Please try to find the earliest case where you find an error and if you made changes to the configGway or configNode.h files that we should know about please include them here as well.

Thanks,

Maarten

platenspeler, 2020-03-01:
As a short guideline: 
- Copy the PlatformIO "src" subdir tp "ESP-sc-gway" sub directory of Arduino IDE and 
- copy "lib" subdir to "library"

Maarten