things4u/ESP-1ch-Gateway #21: ESP-1ch-Gateway not yet working with TTGO T-Beam V1.0

Seeelefant, 2020-02-23:
Hi Maarten,
as already discussed I have a TTGO T-Beam V1.0 which needs a different setup, especially when it comes to the Power Management IC AXP192:

T-Beam version V1.0 official Pinout schema (according gto http://www.lilygo.cn/claprod_view.aspx?TypeId=62&Id=1224&FId=t28:62:28):
[image]

Pins | T_BEAM(AXP192) | T_BEAM_V07
-- | -- | --
LORA_SCK | 5 | 5
LORA_MISO | 19 | 19
LORA_MOSI | 27 | 27
LORA_SS | 18 | 18
LORA_DI0 | 26 | 26
LORA_RST | 23 | 23
GPS_RX_PIN | 34 | 12
GPS_TX_PIN | 12 | 15
I2C_SDA | 21 | 21
I2C_SCL | 22 | 22
PMU_IRQ | 35 | N/A
USER BUTTON | 38 | 39
GPS_PPS | 37 | N/A

Modules | T_BEAM_V10 | T_BEAM_V07
-- | -- | --
GPS | LDO3 | No support
LORA | LDO2 | No support
OLED | DCDC1 | No support

A sample of that code can be found here:
https://github.com/lewisxhe/TTGO-T-Beam/blob/master/TTGO-T-Beam.ino

Best
--Wolfgang

platenspeler, 2020-02-23:
Hi Wolfgang.

So what does not work? You sent me pin-out's but it does not explain whet is not working. And I cannot see your hardware or buy every piece of hardware to find out what is not working,

So please try to explain what your setup does NOT do and why it is failing and what the messages are, I do not know whether the gateway works, or whether part of the sensors work or what is the problem. Without better descriptions I cannot help.

You can set your own pin-out like I did in loraModem.h and if so use pin_out 5.
It is possible that you need to solder a separated line like I did to get the interrupts working.
The T-beam had a pin-out which for LORA is the same as _PIN_OUT==4

- SCK=pin5
- MISO=pin 19
- MOSI=pin 27
- SS/CS = pin 18
- RST = pin14

With this, the gateway will normally work. These are at this moment the only pin's that I need.  If the output of the several sensors is different, Only the sensors might need other settings.
For exaple. the GPS is different and should be something like.

GPS_RX 12
GPS_TX 34

regards,

Maarten

Seeelefant, 2020-02-23:
Well Maarten,
I had to learn that the T-Beam V1 with power management chip AXP192 behaves a little different. If I take your original code nothing is working, every unit must be first initialized via the AXP192 ("switched on"):
````
#include <axp20x.h>

AXP20X_Class pmu;
void AXP192_power(bool on) {
  if (on) {
    pmu.setPowerOutPut(AXP192_LDO2, AXP202_ON);  // Lora on T-Beam V1.0
    pmu.setPowerOutPut(AXP192_LDO3, AXP202_ON);  // Gps on T-Beam V1.0
    pmu.setPowerOutPut(AXP192_DCDC1, AXP202_ON); // OLED on T-Beam v1.0
    // pmu.setChgLEDMode(AXP20X_LED_LOW_LEVEL);
    pmu.setChgLEDMode(AXP20X_LED_BLINK_1HZ);
  } else {
    pmu.setChgLEDMode(AXP20X_LED_OFF);
    pmu.setPowerOutPut(AXP192_DCDC1, AXP202_OFF);
    pmu.setPowerOutPut(AXP192_LDO3, AXP202_OFF);
    pmu.setPowerOutPut(AXP192_LDO2, AXP202_OFF);
  }
}
void AXP192_init() {      
    Wire.begin(21, 22);
    if (pmu.begin(Wire, AXP192_SLAVE_ADDRESS))  {
      Serial.println("AXP192 PMU initialization failed");
    }
    else {
Serial.println("Initialzation...");
      // configure AXP192
      pmu.setDCDC1Voltage(3500);              // for external OLED display
      pmu.setTimeOutShutdown(false);          // no automatic shutdown
      pmu.setTSmode(AXP_TS_PIN_MODE_DISABLE); // TS pin mode off to save power

      // switch ADCs on
      pmu.adc1Enable(AXP202_BATT_VOL_ADC1, true);
      pmu.adc1Enable(AXP202_BATT_CUR_ADC1, true);
      pmu.adc1Enable(AXP202_VBUS_VOL_ADC1, true);
      pmu.adc1Enable(AXP202_VBUS_CUR_ADC1, true);

      // switch power rails on
      AXP192_power(true);

      Serial.println("AXP192 PMU initialized");
    }
}
````
followed by
````
AXP192_init();
AXP192_power(true);

````
in the init section.

The AXP192 also manages sleep modes and charging of the batteries, you can find code samples here:
https://github.com/lewisxhe/TTGO-T-Beam/blob/master/TTGO-T-Beam.ino

Best
--Wolfgang

Seeelefant, 2020-03-21:
Hi Maarten,
the power management code was moved to https://github.com/lewisxhe/AXP202X_Library
Best
--Wolfgang

MohammedBENHADINE, 2022-03-23:
Any news on this one ?