things4u/ESP-1ch-Gateway #22: WDT resetting after connected to SSID.

cirujadigital, 2020-02-27:
Im getting a constant reset after the I get connected to the SSID. My setup is NODEMCU 1.0 LOLIN + RFM95W, got SPIFFS enabled,  No GPS  or OLED attached (they are disabled), wifimanager as well.

I set the NTP server to my country's one: ar.pool.ntp.org

```
Do Asserts
.CH=	0
.SF=	7
.FCNT=	0
.DEBUG=	1
.PDEBUG=	96
.CAD=	1
.HOP=	0
.NODE=	0
.BOOTS=	0
.RESETS=	0
.WIFIS=	1
.VIEWS=	0
.REFR=	0
.REENTS=	0
.NTPETIM=	0
.NTPERR=	0
.NTPS=	0
.FILEREC=	0
.FILENO=	0
.FILENUM=	0
.DELAY=	0
.TRUSTED=	1
.EXPERT=	0
.SEEN=	1
.MONITOR=	1
readGwayCfg:: return OK
MAC: 60:01:94:3c:86:34, len=17
Host=************WiFi Connected to WayFay2 on IP=192.168.43.60
Local UDP port=1700
UDP Connection successful
Gateway ID: ***************, Listening at SF7 on 903.90 MHz.
resolveHost:: OK=ar.pool.ntp.org IP=168.96.251.197

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
ets_main.c 
```

platenspeler, 2020-02-27:
Hi,

What code version are you using (configGway.h)?
Sometimes setting the FORMAT to 1 could help to init SPIFFS as well and prevent reboots.
Did you switch wifimanager off or on? Software works not ok with ESP32.

Maarten

cirujadigital, 2020-02-27:
> Hi,
> 
> What code version are you using (configGway.h)?
> Sometimes setting the FORMAT to 1 could help to init SPIFFS as well and prevent reboots.
> Did you switch wifimanager off or on? Software works not ok with ESP32.
> 
> Maarten

Hi Maarten, I tried v5 and the last v6, all using ESP8266 (NodeMCU) and RFM95W.  I changed SPIFFS FORMAT to 1, but still issues persist -I already checked my connections, this is what i've got:
```
WWW Server started on port 80
Unknown transceiver=FF, pins.rst =2, pins.ss  =0, pins.dio0 =5, pins.dio1 =5, pins.dio2 =4

Abort called

>>>stack>>>

ctx: cont
sp: 3ffffcf0 end: 3fffffc0 offset: 01b0
3ffffea0:  00000000 00000000 ffff277c 4021c108  
3ffffeb0:  000000ff 3ffe859f 3fff277c 4020652c  
3ffffec0:  00000000 00000000 00000064 4021f487  
3ffffed0:  3fff254c 3fff2070 3fff1f30 402112f4  
3ffffee0:  00000086 00000034 feefeffe feefeffe  
3ffffef0:  303a3036 34393a31 3a63333a 333a3638  
3fffff00:  fe000034 38707365 2d363632 36386333  
3fffff10:  00000000 00000000 ffefeffe 4022d0b0  
3fffff20:  2e52ebbf ff2e003d 00000000 00000000  
3fffff30:  ffefef00 00000000 00000000 ffefeffe  
3fffff40:  00000000 00000000 ffefeffe 00000000  
3fffff50:  00000000 ffefeffe 00000000 00000000  
3fffff60:  ffefeffe feefeffe feefeffe feefeffe  
3fffff70:  00000000 00000000 ffefeffe feefeffe  
3fffff80:  00000000 feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3fff28a0  
3fffffa0:  3fffdad0 00000000 3fff286c 4021e7c0  
3fffffb0:  feefeffe feefeffe 3ffe899c 401011e5  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
```

Thanks in advance.

platenspeler, 2020-02-28:
The erro rcoming from the board is:

Unknown transceiver=FF, pins.rst =2, pins.ss  =0, pins.dio0 =5, pins.dio1 =5, pins.dio2 =4

Normally this means a wiring problem. dio0 and dio1 are connected to GPIO 5 and dio2 to GPIO 4 which is strange (but should not matter). The function initLoraModem() says that neither sx1276 or sx1272 is found.

Can you send me the loramModem.h file (and what pin-out you have selected/defined in condifgGway.h?). Maybe I can find out something.

Ik you have to assigne separate GPIO pins to the dio pins of the RFP95, please select at least dio0 and dio1. dio2 is less important.

Maarten

cirujadigital, 2020-02-29:
Im using the default pinout in configGway.h (PIN_OUT=1), didn't changed anything from loraModem.h or configGway.h

What its really strange is that after enabling the WifiManager option, and selecting my SSID form the AP mode, the ESP8266 connects to it BUT when I try to enter to the webserver via browser, I get the WifiManager display page (in theory, I should  not be seeing this)

platenspeler, 2020-02-29:
By the way, I do not have a version 6.0 anymore. The latest version is without the number.

Maarten

platenspeler, 2020-02-29:
The is from the loraModem.h file. As you can see you setting is not _PIN_OUT==1
You can add _PINT_OUT==5 to reflact the pin out that you are using for your board.

Maarten

#if _PIN_OUT==1
// ----------------------------------------------------------------------------
// Definition of the GPIO pins used by the Gateway for Hallard type boards
//
struct pins {
	uint8_t dio0=15;	// GPIO15 / D8. For the Hallard board shared between DIO0/DIO1/DIO2
	uint8_t dio1=15;	// GPIO15 / D8. Used for CAD, may or not be shared with DIO0
	uint8_t dio2=15;	// GPIO15 / D8. Used for frequency hopping, don't care
	uint8_t ss=16;		// GPIO16 / D0. Select pin connected to GPIO16 / D0
	uint8_t rst=0;		// GPIO 0 / D3. Reset pin not used	
	// MISO 12 / D6
	// MOSI 13 / D7
	// CLK  14 / D5
} pins;