things4u/ESP-1ch-Gateway #23: Setting #define OLED 0 breaks compilation

tavdog, 2020-02-29:
If I set #define OLED 0 it won't compile with following error message :
```
SP-sc-gway:305: error: 'msg_oLED' was not declared in this scope
   msg_oLED("FORMAT");
                    ^
ESP-sc-gway:367: error: 'msg_oLED' was not declared in this scope
  msg_oLED("WIFI STA");
                     ^
ESP-sc-gway:484: error: 'msg_lLED' was not declared in this scope
  msg_lLED("GET TIME",".");
                         ^
/Users/tavis/code/arduino-projects/ESP-1ch-Gateway/ESP-sc-gway/_wwwServer.ino: In lambda function:
_wwwServer:1300: error: 'msg_oLED' was not declared in this scope
   msg_oLED("FORMAT");  
```

platenspeler, 2020-02-29:
I could have made a mistake in earlier version, but I think in any case the OLED definition is correct in the latest version. What version of Gateway are you using? It is shown iin configGway.h file.

Maarten

tavdog, 2020-02-29:
I used V.6.2.0.E.EU868; 200129b because the Arduino app complains about the changes made for platformIO.

platenspeler, 2020-02-29:
Ah. You can download the 6.2.3 version and copy the content of the src directory of PlatformIO to the ESP-sc-gway directory, and the content of the lib directory of PlatformIO to the libraries directory.

Maarten

tavdog, 2020-03-01:
That works, thank you.

platenspeler, 2020-03-01:
Good to hear, Thank you!

closed 2020-03-01 by platenspeler