things4u/ESP-1ch-Gateway #24: us-west.thethings.network not working

tavdog, 2020-03-01:
I got everything working on the eu router but when I change to  us-west.thethings.network the "Last Seen" field on the TTN console page never updates.  This might not be a problem with this firmware but I just thought I'd see if you had any ideas.  My gateway is at https://console.thethingsnetwork.org/gateways/eui-b4e62dffff85ccd1

platenspeler, 2020-03-01:
I do not have permission to see your gateway. Could you mail me a screenshot of the GUI and indicate which line is giving you problems?

Thanks,

Maarten

BTW. For some reasons the first line of the "seen" array gets corrupted. I will have a second or third look at it :-)

tavdog, 2020-03-01:
First image is just to show configuration of us-west.thethings.network. 
[image]
Second image shows last seen is not getting updated.  When using route.eu it updates every 2 minutes. But with us-west it never updates.
[image]

platenspeler, 2020-03-01:
Please have a look at the sensors you have as well in the TTN console.
If all right, not only should the gateway be seen, but the nodes should also connect to your gateway.

Inf the Message History section of the GUI does it show the sensors connected to your gateway alright?

Sometimes the gateway does not behave alright in which case I use "FORMAT" and then "RESET" to reset the whole thing. The SPIFFS filesystem does not always behave well.

Maarten

tavdog, 2020-03-01:
I don't have any nodes connecting at this time. I mainly wanted to setup a public gateway for my town.  I'll get a node connecting and then try us-west, maybe "last seen" is handled differently on the us-west router.  thanks for all the work on this project. Feel free to close this issue.

platenspeler, 2020-03-01:
lastSeen is designed to only report active nodes. So if there are no nodes connected then it should not report anythings. Only like I said it does resport a dummy node the first time.

Maarten

platenspeler, 2020-03-01:
So it could be that the gateway is working fine for you. Interested to hear your experiences

closed 2020-03-01 by platenspeler