things4u/ESP-1ch-Gateway #27: ESP32 wemos with hand soldered lora module

danielkoek, 2020-03-08:
Hey, First of all, awesome work you did here!
I am not sure what is going on cause i can't seem to really debug it well. 
Soldered a little gateway together by hand using the wiring of the sparkfuns lora 1ch gateway schemetics, only leaving out the RST pin since there is no need for it. Checked the wiring 3 times but can't seem to get it working, maybe i am missing something but i keep getting a abort, here is the output from the Serial:

entry 0x400806ac
SPIFFS begin
Format SPIFFS Filesystem Done
Do Asserts
readConfig ERR:: file=/gwayConfig.txt does not exist ..
readConfig:: Error reading config file
setup:: readGwayCfg: ERROR readCfgCfg Failed
MAC: 30:ae:a4:37:e6:1c, len=17
0:0:1. WlanConnect SSID=Boogmakerstraat 45
WlanStatus:: NO SSID
..1:0:1. WlanConnect SSID=Boogmakerstraat 45
WlanStatus:: CONNECTED ssid=Boogmakerstraat 45
Host=esp32-37e61c WiFi Connected to Boogmakerstraat 45 on IP=192.168.1.108
Local UDP port=1700
UDP Connection successful
Gateway ID: 30aea4ffff37e61c, Listening at SF9 on 868.10 MHz.
resolveHost:: OK=nl.pool.ntp.org IP=94.228.143.152
Time set=Sunday 8-3-2020 17:39:53
resolveHost:: OK=router.eu.thethings.network IP=52.169.76.203
setupOta:: Started
Ready IP address: 192.168.1.108
WARNING:: readSeen, history file not exists /gwayNum.txt
WWW Server started on port 80
Unknown transceiver=0, pins.rst =14, pins.ss  =18, pins.dio0 =26, pins.dio1 =33, pins.dio2 =32

abort() was called at PC 0x400dea6c on core 1

Backtrace: 0x4008c468:0x3ffb1e40 0x4008c699:0x3ffb1e60 0x400dea6c:0x3ffb1e80 0x400dec15:0x3ffb1eb0 0x400e2dfd:0x3ffb1ee0 0x400ecd97:0x3ffb1fb0 0x40088bb1:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
SPIFFS begin
Format SPIFFS Filesystem Done
Do Asserts
readConfig ERR:: file=/gwayConfig.txt does not exist ..

If you need any more info please let me know, would really like this to work!

Kind regards, 
Daniël

platenspeler, 2020-03-10:
Can you post a picture of the pin-out? Based on the output your board is not recognized by the software. You use _PIN_OUT==4 which is standard for ESP32 boards. Is it also standard for your board?

Unknown transceiver=0, pins.rst =14, pins.ss =18, pins.dio0 =26, pins.dio1 =33, pins.dio2 =32

The Sparfun node may be different.

Maarten

danielkoek, 2020-03-10:
It would be easier to write them all down:

DIO2:32
Dio1:33
Dio0:26
Vdd:3v
Dio4:NC
Dio3:NC
Gnd:NC 
Ant: connected to the wire

Then the second row

Gnd:NC
Miso:miso/19
Mosi:mosi/23
Sck:18
Nss:16
RST:NC
Dio5:NC
Gnd:gnd

Maybe I am missing something but that is the pinout as well as specified right or do I have to connect rst or maybe I messed up nss and sck that nss should the ss pin so change the software to 16 instead of 18? Will check that out the next day, don't have my laptop at hand atm

danielkoek, 2020-03-16:
Changed the FW now too, but it didn't change anything expect the pinout is now 
Unknown transceiver=0, pins.rst =27, pins.ss  =16, pins.dio0 =26, pins.dio1 =33, pins.dio2 =32

platenspeler, 2020-03-18:
I see now that the software thinks that rst is 27. Did you change the software to be in line with your hardware? 
Make sure that you change the software in loraModem.h (and pin definition in configGway.h) so that these are the same as the hardware.

Maarten

danielkoek, 2020-03-19:
Yea that is right, but i didn't connect rst to anything now, so it is not a pin used, other then that i just set everything to esp32 as far as i know, turned off the oled and all that kind of stuff, so i am really scratching my head what is going on, any suggestion on how i could see why it is failing?

Daniël

platenspeler, 2020-03-28:
I will have  last look. but since all of my Gateway work, I am not sure what is the problem with yours.

platenspeler, 2020-03-28:
Perhaps send me a version of your configNode.h and configGway.h fil together with a picyure of your pin-out of the node.

Maarten

danielkoek, 2020-03-31:
Yea ofc no bother at all,  here you go: 
[image]
[image]

danielkoek, 2020-03-31:
```
// 1-channel LoRa Gateway for ESP32 and ESP8266
// Copyright (c) Maarten Westenberg 2016-2020

#define VERSION "V.6.2.3.E.EU868; PlatformIO 200223a"
//
// Based on work done by Thomas Telkamp for Raspberry PI 1ch gateway and many others.
//
// All rights reserved. This program and the accompanying materials
// are made available under the terms of the MIT License
// which accompanies this distribution, and is available at
// https://opensource.org/licenses/mit-license.php
//
// NO WARRANTY OF ANY KIND IS PROVIDED
//
// Author: Maarten Westenberg (mw12554@hotmail.com)
//
// This file contains a number of compile-time settings that can be set on (=1) or off (=0)
// The disadvantage of compile time is minor compared to the memory gain of not having
// too much code compiled and loaded on your ESP device.
//
// NOTE:
// If version is for ESP32 Heltec board, compile with ESP32 setting and board
// "ESP32 Dev Module" or "Heltec WiFi Lora 32"
//
// For ESP8266 Wemos: compile with "Wemos R1 D1" and choose
// the right _PIN_OUT below. Selecting OLED while that is not connected does not
// really matter (so you can leave that in).
//
// The source has been optimized for PlatformIO usage. Than means the some variables
// can be define in the .platformio.ini file. Pleas look at the example file to see
// how to set these #defines
//
// ========================================================================================

// Define whether we should do a formatting of SPIFFS when starting the gateway
// This is usually a good idea if the webserver is interrupted halfway a writing
// operation. Also to be used when software is upgraded
// Normally, value 0 is a good default and should not be changed.
#if !defined _SPIFFS_FORMAT
#define _SPIFFS_FORMAT 1
#endif

// Allows configuration through WifiManager AP setup. Must be 0 or 1
#if !defined _WIFIMANAGER
#define _WIFIMANAGER 0
#endif

// Debug message will be put on Serial is this one is set.
// If set to 0, no printing to USB devices is done.
// Set to 1 it will print all user level messages (with correct debug set)
// If set to 2 it will also print interrupt messages (not recommended)
#if !defined _DUSB
#define _DUSB 2
#endif

// Define the monitor screen. When it is greater than 0 then logging is displayed in
// the special screen at the GUI.
// If _DUSB is also set to 1 then most messages will also be copied to USB devices.
#define _MONITOR 1

// Gather statistics on sensor and Wifi status
// 0= No statistics
// 1= Keep track of messages statistics, number determined by _MAXSTAT
// 2= Option 1 + Keep track of messages received PER each SF (default)
// 3= See Option 2, but with extra channel info (Not used when Hopping is not selected)
#define _STATISTICS 3

// Define the frequency band the gateway will listen on. Valid options are
// EU863_870	Europe
[... 196 lines cut ...]
//	- Last seen 'seconds since 1/1/1970'
//	- SF seen (8-bit integer with SF per bit)
// The initial version _NUMMAX stores max this many nodes, please make
// _MAXSEEN==0 when not used
#define _MAXSEEN 20
#define _SEENFILE "/gwayNum.txt"

// Name of he configfile in SPIFFs	filesystem
// In this file we store the configuration and other relevant info that should
// survive a reboot of the gateway
#define CONFIGFILE "/gwayConfig.txt"

// Define the correct radio type that you are using
#define CFG_sx1276_radio
//#define CFG_sx1272_radio

// Serial Port speed
#define _BAUDRATE 115200 // Works for debug messages to serial momitor

// MQTT definitions, these settings should be standard for TTN
// and need no changing
#define _TTNSERVER "router.eu.thethings.network"
#define _TTNPORT 1700 // Standard port for TTN

```

danielkoek, 2020-03-31:
```
// sensor.h; 1-channel LoRa Gateway for ESP8266
// Copyright (c) 2016-2020 Maarten Westenberg version for ESP8266
//
// based on work done by Thomas Telkamp for Raspberry PI 1ch gateway
// and many other contributors.
//
// All rights reserved. This program and the accompanying materials
// are made available under the terms of the MIT License
// which accompanies this distribution, and is available at
// https://opensource.org/licenses/mit-license.php
//
// NO WARRANTY OF ANY KIND IS PROVIDED
//
// Author: Maarten Westenberg (mw12554@hotmail.com)

//
// This file contains a number of compile-time settings and declarations that are'
// specific to the LoRa rfm95, sx1276, sx1272 radio of the gateway.
//
//
// ------------------------------------------------------------------------------------

// It is possible to use the gateway as a normal sensor node also. In this case,
// substitute the node info below.
#if _GATEWAYNODE == 1

// Valid coding for internal sensors are LCODE and RAW.
// Make sure to only select one.
//#define _LCODE 1
#define _RAW 1
#define _CHECK_MIC 1
#define _SENSOR_INTERVAL 300

#define _DEVADDR               \
	{                          \
		0xAA, 0xAA, 0xAA, 0xAA \
	}
#define _APPSKEY                                                                                       \
	{                                                                                                  \
		0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB \
	}
#define _NWKSKEY                                                                                       \
	{                                                                                                  \
		0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC \
	}
#define _SENSOR_INTERVAL 300
// For ESP32 based T_BEAM/TTGO boards these two are normally included
// If included make value 1, else if not, make them 0
#define _GPS 1
#define _BATTERY 1

#endif //_GATEWAYNODE

#if _TRUSTED_NODES >= 1
struct nodex
{
	uint32_t id; // This is the LoRa ID (coded in 4 bytes uint32_t
	char nm[32]; // Name of the node
};

// Add all your named and trusted nodes to this list
nodex nodes[] = {
	{0x260116BD, "lora-34 PIR node"},		  // F=0
	{0x26011152, "lora-35 temp+humi node"},   // F=0
	{0x2601148C, "lora-36 test node"},		  // F=0
	{0x00000000, "lora-00 well known sensor"} // F=0
};

#endif //_TRUSTED_NODES

// In some cases we like to decode the lora message at the single channel gateway.
// In thisase, we need the NkwSKey and the AppsSKey of the node so that we can decode
// its messages.
// Although this is probably overkill in normal gateway situations, it greatly helps
// in debugging the node messages before they reach the TTN severs.
//
#if _LOCALSERVER == 1
struct codex
{
	uint32_t id;		  // This is the device ID (coded in 4 bytes uint32_t
	unsigned char nm[32]; // A name string which is free to choose
	uint8_t nwkKey[16];   // The Network Session Key of 16 bytes
	uint8_t appKey[16];   // The Application Session Key of 16 bytes
};

// Sometimes we want to decode the sensor completely as we do in the TTN server
// This means that for all nodes we want to view the data of, we need to provide
// the AppsSKey and the NwkSKey

// Definition of all nodes that we want to decode locally on the gateway.
//
codex decodes[] = {
	{0xAAAAAAAA, "lora-EE", // F=0
	 {0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB},
	 {0xDD, 0xDD, 0xDD, 0xDD, 0xDD, 0xDD, 0xDD, 0xDD, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC}},
	{0xBBBBBBBB, "lora-FF", // F=0
	 {0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB, 0xBB},
	 {0xDD, 0xDD, 0xDD, 0xDD, 0xDD, 0xDD, 0xDD, 0xDD, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC}},
	{0x00000000, "lora-00", // F=0
	 {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00},
	 {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}}};
#endif //_LOCALSERVER

// If you have a second back-end server defined such as Semtech or loriot.io
// your can define _THINGPORT and _THINGSERVER with your own value.
// If not, make sure that you do not define these, which will save CPU time
// Port is UDP port in this program
//
// Default for testing: Switched off
// #define _THINGSERVER "your.extra.server"		// Server URL of the LoRa-udp.js handler
// #define _THINGPORT 1700						// Port 1700 is old compatibility

// Wifi definitions
// WPA is an array with SSID and password records. Set WPA size to number of entries in array
// When using the WiFiManager, we will overwrite the first entry with the
// accesspoint we last connected to with WifiManager
// NOTE: Structure needs at least one (empty) entry.
//		So WPASIZE must be >= 1
struct wpas
{
	char login[32]; // Maximum Buffer Size (and allocated memory)
	char passw[64];
};

// Please fill in at least ONE valid SSID and password from your own WiFI network
// below. This is needed to get the gateway working
//
wpas wpa[] = {
	{"Boogmakerstraat 45", "danielandclaire"}};

// Define the name of the accesspoint if the gateway is in accesspoint mode (is
// getting WiFi SSID and password using WiFiManager)
#define AP_NAME "ESP32-Gway-Things4U"
#define AP_PASSWD "0297282242"

// For asserting and testing the following defines are used.
//
#if !defined(CFG_noassert)
#define ASSERT(cond) \
	if (!(cond))     \
	gway_failed(__FILE__, __LINE__)
#else
#define ASSERT(cond) /**/
#endif

```

danielkoek, 2020-03-31:
Thanks in advance for helping me out with this, if this doesn't work i give up with a handmade one and will just order a prefab

platenspeler, 2020-04-12:
I cannot find anything wrong with it. Can you send the loraModem.h file as well, I like to see your definitions of _PIN_OUT == 5.

danielkoek, 2020-04-14:
I am really a big idiot, i changed the pins but not the defines eg:
#define SCK 18
#define MISO 19
#define MOSI 23
#define SS 16
#define DIO0 26

Is the correct one, but i had the old one from pinout 4, left in place, that obviously doesn't work!
thank you so so much for all your help with this! now i can get started with automating my community garden!

closed 2020-04-14 by danielkoek