things4u/ESP-1ch-Gateway #29: Contant reboot when webserver on port 80

rk125, 2020-04-09:
Hello all,
In case someone experiences constant gateway reboots and scratches heads why did it work before and not any more after reflashing gateway.
Gateway is crashing with no obvious reason after connecting to wifi and starting web server. Reset reason did not give any hint. Looks like this:
"
SPIFFS begin
Do Asserts
.CH=    0
.SF=    9
.FCNT=  0
.DEBUG= 1
.PDEBUG=        96
.CAD=   1
.HOP=   0
.NODE=  0
.BOOTS= 0
.RESETS=        0
.WIFIS= 11
.VIEWS= 0
.REFR=  0
.REENTS=        0
.NTPETIM=       0
.NTPERR=        0
.NTPS=  11
.FILEREC=       0
.FILENO=        0
.FILENUM=       0
.DELAY= 0
.TRUSTED=       1
.EXPERT=        0
.SEEN=  1
.MONITOR=       1
readGwayCfg:: return OK
MAC: 24:0a:c4:a6:1b:ec, len=17
0:0:1. WlanConnect SSID=i8ny
WlanStatus:: CONNECTED ssid=i8ny
Host=esp32-a61bec WiFi Connected to i8ny on IP=192.168.1.144
Local UDP port=1700
UDP Connection successful
Gateway ID: 240ac4ffffa61bec, Listening at SF9 on 868.10 MHz.
resolveHost:: OK=time.b41 IP=192.168.10.10
Time set=Monday 6-4-2020 20:23:37
resolveHost:: OK=loraserver.b41 IP=192.168.10.20
setupOta:: Started
Ready IP address: 192.168.1.144
readSeen:: No more info left in file, i=4
WWW Server started on port 80
OLED_ADDR=0x3C
 --- Setup() ended, Starting loop() ---
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8
SPIFFS begin
"
Some other board was giving meditation error, but that did not give any hint either. Code was perfectly working before.
2 ways to solve it:
- disable web server, though not useful
- specify web server port other than 80, I used 88. Then you connect to gw with http://<ip>:<port> of course

No idea why it was not showing up before (consistently, I was able to get it working before), but seems to me it is kinda race condition, if other process starts on port 80 before web server, gateway crashes when web server starts on same port. Same code was working for me before and once I already stumbled on this behaviour, but after changing gw board, it started to work, I blamed board back then. Now, after last recompilation, I could not get it running at all with any board.
Crazy thing is that it shows up on any version of gateway code. I tried with esp32, esp8266, changed esp platform versions. Arduino, platform.io - does not matter, same crash.

Reggie

platenspeler, 2020-04-11:
Dear Reggie,

You probably known that debugging exceptions with the ESP8266 and ESP32 is difficult.
If it helpt, I'm working on a new subversion which is faster and has new/improved connection properties.
My goals is to get downstream working correctly and that's the reasons why I am working on this version.

Maarten

rk125, 2020-04-12:
Hello Maarten,
No worrries! Main think that there is workaround! I just put here if someone else steps on same issue.
Thanks for the great gateway ;)
Reggie

closed 2020-04-12 by rk125