things4u/ESP-1ch-Gateway #33: Wemos D1-mini board esp8266 can not work

Supowang1989, 2020-05-12:
I download the code to Wemos D1-mini board, get the wrong log in serial port as follows:
;$\0l恟$鈔?l?b|帎抮b?b騨nlnnbbpp?blrl
Can someone help me to solve this problem？ 
thanks！

platenspeler, 2020-05-13:
Hi,

Do you really get those sybols as described in the email?

Supowang1989, 2020-05-14:
> Hi,
> 
> Do you really get those sybols as described in the email?

Yes, the baud rate I use is 115200，_PIN_OUT set  2，board is [https://oshpark.com/shared_projects/3Cg38FFD](url)
In Ardunio ，I chose the LOLIN（Wemos）D1 mini Lite board

platenspeler, 2020-05-16:
I am looking at the board at oshpark. However, it does not seem to be the same as the Hallard board I used and described at PIN out 2. If it is not the same board, chances are that the board uses other pins as well.

What are the pin connections of this board (DIO0, DIO1, RST, )? If we know these, than we can make a PIN definition.

It could be that you are looking at the wrong pins and this produces strange output.

Maarten