things4u/ESP-1ch-Gateway #34: ack_lost, uplinks_missed and uplinks_lost with lorawan-server

MihaMarkic, 2020-05-14:
Hi guys,

I'm seeing a lot of ack_lost, uplinks_missed and uplinks_lost events with [lorawan_server](https://github.com/gotthardp/lorawan-server).
Both gateway and server are located in the same room on the same network. How do I even start to figure out why the errors?
Note: the data seem coming through (from device to server via gateway).

I've set debug to 3 via web ui and looking at serial monitor I don't see any obvious problem.
Typical output is below:

> Thu 14-05-2020 17:04:27- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=592.976816
> Thu 14-05-2020 17:04:27- loop:: readUdp available
> Thu 14-05-2020 17:04:27- Dwn readUdp:: message ident=4
> Thu 14-05-2020 17:04:27- Dwn readUdp PKT_PULL_ACK: micr=593.493082, size=4 From 10.9.9.5, port 1680
> , data: 1:F:BD:4:
> Thu 14-05-2020 17:04:27- Dwn readUdp:: ret=4, data=1 15 189 4
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- S_RX no INTR:: 0
> Thu 14-05-2020 17:04:28- RX HEADER:: 16
> Thu 14-05-2020 17:04:28- RX HEADER:: 16
> Thu 14-05-2020 17:04:28- RX HEADER:: 16
> Thu 14-05-2020 17:04:28- RX HEADER:: 16
> Thu 14-05-2020 17:04:28- RX HEADER:: 16
> Thu 14-05-2020 17:04:28- RX HEADER:: 16
> Thu 14-05-2020 17:04:28- RX HEADER:: 16
> Thu 14-05-2020 17:04:28- RX HEADER:: 16
> Thu 14-05-2020 17:04:28- RX HEADER:: 16
> Thu 14-05-2020 17:04:28- RX HEADER:: 16
> Thu 14-05-2020 17:04:28- RX HEADER:: 16
> Thu 14-05-2020 17:04:28- RX HEADER:: 16
> Thu 14-05-2020 17:04:28- RX HEADER:: 16
> Thu 14-05-2020 17:04:28- RX HEADER:: 16
[... 131 lines cut ...]
> Thu 14-05-2020 17:05:28- Send Pushdata sendstat
> Thu 14-05-2020 17:05:28- loop:: readUdp available
> Thu 14-05-2020 17:05:28- Dwn readUdp:: message ident=1
> Thu 14-05-2020 17:05:28- Dwn PKT_PUSH_ACK:: size=4, From 10.9.9.5
> Thu 14-05-2020 17:05:28- Dwn readUdp:: ret=4, data=1 60 168 1
> Thu 14-05-2020 17:05:30- SCAN:: I=CDDONE CDDETD , F=0, SF=7, E=0, S=RX   , eT=6700312, dT=2376
> Thu 14-05-2020 17:05:30- S_RX no INTR:: 0
> Thu 14-05-2020 17:05:30- S_RX no INTR:: 0
> Thu 14-05-2020 17:05:30- S_RX no INTR:: 0
> Thu 14-05-2020 17:05:30- S_RX no INTR:: 0
> Thu 14-05-2020 17:05:30- RXTOUT:: I=RXTOUT , F=0, SF=7, E=0, S=RX   , eT=6713261, dT=15290
> Thu 14-05-2020 17:05:47- sendUdp: WlanConnect connected to=Partizan. Server IP=10.9.9.103
> Thu 14-05-2020 17:05:47- M PKT_PULL_DATA request, len=12
> 1:35:E5:2:24:6F:28:FF:FF:B1:BF:84:
> Thu 14-05-2020 17:05:47- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=672.976849
> Thu 14-05-2020 17:05:47- loop:: readUdp available
> Thu 14-05-2020 17:05:47- Dwn readUdp:: message ident=4
> Thu 14-05-2020 17:05:47- Dwn readUdp PKT_PULL_ACK: micr=672.993680, size=4 From 10.9.9.5, port 1680
> , data: 1:35:E5:4:
> Thu 14-05-2020 17:05:47- Dwn readUdp:: ret=4, data=1 53 229 4
> Thu 14-05-2020 17:05:49- SCAN:: I=CDDONE CDDETD , F=0, SF=7, E=0, S=RX   , eT=18823241, dT=2351
> Thu 14-05-2020 17:05:49- S_RX no INTR:: 0
> Thu 14-05-2020 17:05:49- S_RX no INTR:: 0
> Thu 14-05-2020 17:05:49- S_RX no INTR:: 0
> Thu 14-05-2020 17:05:49- RXTOUT:: I=RXTOUT , F=0, SF=7, E=0, S=RX   , eT=18832644, dT=11718

platenspeler, 2020-05-15:
Better not use debug levels larger than 1, since the timing of the gateway becomes influenced.
Sometimes use 2 if yo are debugging bu normally not level 3.

What version of the gateway are you using (listed on the top of the webpage)? And are you using Arduino IDE or PlatformIO? What debug patterns did you select on the GUI?

Maarten

MihaMarkic, 2020-05-15:
OK, no more level > 1.
It's _Version: V.6.2.4.EU868; PlatformIO 200428n_ with PlatformIO (obviously). For the give listing I used level 2 with almost everything except SCN and possible PRE I think.

platenspeler, 2020-05-16:
OK, 

Try debug==level 1, and select RX and MAIN. Can you send a small part of the output to Serial?

MihaMarkic, 2020-05-19:
This is on gateway (write failures happen when I change debug settings, shouldn't matter I guess)
```
[E][WiFiClient.cpp:392] write(): fail on fd 56, errno: 104, "Connection reset by peer"
[E][WiFiClient.cpp:392] write(): fail on fd 58, errno: 104, "Connection reset by peer"
[E][WiFiClient.cpp:392] write(): fail on fd 60, errno: 104, "Connection reset by peer"
Tue 19-05-2020 15:03:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=76.48058
Tue 19-05-2020 15:04:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=96.49193
Tue 19-05-2020 15:04:35- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=116.49228
Tue 19-05-2020 15:04:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=136.49599
Tue 19-05-2020 15:05:05- UP receivePkt:: rxPkt: t=Tue 19-05-2020 15:05:05, f=0, sf=7, a=d0:02:ef:4e:, flags=50, addr=0, len=23
Tue 19-05-2020 15:05:05- UP receivePacket:: start: micr=146.237306, tmst=0.0
Tue 19-05-2020 15:05:05- UP receivePacket:: sendUdp: micr=146.246412, tmst=146.245311
Tue 19-05-2020 15:05:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=156.49213
Tue 19-05-2020 15:05:35- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=176.49259
Tue 19-05-2020 15:05:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=196.47841
Tue 19-05-2020 15:06:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=216.49028
Tue 19-05-2020 15:06:35- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=236.47945
Tue 19-05-2020 15:06:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=256.47709
Tue 19-05-2020 15:07:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=276.47537
Tue 19-05-2020 15:07:35- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=296.49265
Tue 19-05-2020 15:07:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=316.47777
Tue 19-05-2020 15:08:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=336.49099
Tue 19-05-2020 15:08:35- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=356.49286
Tue 19-05-2020 15:08:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=376.47790
Tue 19-05-2020 15:09:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=396.49036
Tue 19-05-2020 15:09:35- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=416.48068
Tue 19-05-2020 15:09:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=436.47853
Tue 19-05-2020 15:10:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=456.48036
Tue 19-05-2020 15:10:35- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=476.49188
Tue 19-05-2020 15:10:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=496.47828
Tue 19-05-2020 15:11:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=516.47979
Tue 19-05-2020 15:11:35- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=536.48158
Tue 19-05-2020 15:11:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=556.47756
Tue 19-05-2020 15:12:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=576.49168
Tue 19-05-2020 15:12:35- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=596.48585
Tue 19-05-2020 15:12:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=616.49018
Tue 19-05-2020 15:13:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=636.49298
Tue 19-05-2020 15:13:35- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=656.47946
Tue 19-05-2020 15:13:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=676.47799
Tue 19-05-2020 15:14:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=696.47974
Tue 19-05-2020 15:14:35- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=716.49286
Tue 19-05-2020 15:14:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=736.47767
Tue 19-05-2020 15:15:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=756.49046
Tue 19-05-2020 15:15:35- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=776.48108
Tue 19-05-2020 15:15:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=796.47885
Tue 19-05-2020 15:16:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=816.48050
Tue 19-05-2020 15:16:35- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=836.49277
Tue 19-05-2020 15:16:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=856.48950
Tue 19-05-2020 15:17:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=876.47848
Tue 19-05-2020 15:17:35- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=896.48148
Tue 19-05-2020 15:17:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=916.47741
Tue 19-05-2020 15:18:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=936.49283
Tue 19-05-2020 15:18:35- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=956.50291
Tue 19-05-2020 15:18:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=976.47858
Tue 19-05-2020 15:19:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=996.47831
Tue 19-05-2020 15:19:35- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=1016.48755
Tue 19-05-2020 15:19:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=1036.49046
Tue 19-05-2020 15:20:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=1056.49221
Tue 19-05-2020 15:20:35- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=1076.48075
Tue 19-05-2020 15:20:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=1096.49105
Tue 19-05-2020 15:21:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=1116.47870
Tue 19-05-2020 15:21:35- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=1136.49242
Tue 19-05-2020 15:21:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=1156.49057
Tue 19-05-2020 15:22:15- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=1176.47882
[... cut ...]
  | warning | 2020-05-19 15:06:56 | 2020-05-19 15:06:56 | 1 | gateway | 246F28FFFFB1BF84 | ack_lost | 100.0 |  
  | warning | 2020-05-19 15:05:06 | 2020-05-19 15:05:06 | 1 | node | 56ED0CFB | repeated_reset | 1 |  
  | info | 2020-05-19 15:05:06 | 2020-05-19 15:05:06 | 1 | device | 0031E81A65A01F92 | join | <<"56ED0CFB">> |  
  | info | 2020-05-19 15:02:57 | 2020-05-19 15:02:57 | 1 | gateway | 246F28FFFFB1BF84 | connected | {{INTERNAL_IP},1700} |  
  | info | 2020-05-18 17:03:01 | 2020-05-18 17:03:01 | 1 | server | lorawan@77a881b8acfc | started
```

platenspeler, 2020-05-26:
Are you getting real errors? Because I changed the output of the monitor process such that the warnings/output in your file do not occurr in debug==1. And the message included above: Tue "19-05-2020 15:30:55- UP ESP-sc-gway:: PKT_PULL_DATA message sent: micr=1696.49144" is just a message that the gateway did send a PULL_DATA message to the server (but not an error).

Maarten

platenspeler, 2020-05-27:
Ok, the message from the server is more serious.I'll look into it.

antonio-emilio, 2020-06-28:
I'm also getting a lot of messages like this one:
"Thu 14-05-2020 17:04:42- SCAN:: I=CDDONE CDDETD , F=0, SF=7, E=0, S=RX , eT=13414520, dT=2241"
And then, eventually I get the message:
"Thu 14-05-2020 17:04:28- UP RXPK:: {"rxpk":[{"chan":0,"rfch":0,"freq":868.099975,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":12,"rssi":-83,"size":26,"data":"QPsM7VaAMAAB5A2lEgMPpYAoyVTXSMexT2Y=","tmst":594866936}]} , length=209""
It's working, but I don't know why this is happening.