things4u/ESP-1ch-Gateway #35: Could this library use a LoRaWAN node?

IoTThinks, 2020-05-14:
Hi everyone,
This project is excellent for single channel LoRaWan gateway.
I see there is Gateway node mode for the gateway.

Could I make it become a LoRaWAN node?
So the LoRaWAN node (without WiFi) send packets to this LoRaWAN gateway via LoRaWAN.

Thanks a lot.

platenspeler, 2020-05-15:
You can turn the ESP8266 or ESP32 device into a node. You need LMIC6 (preferrably) for that. I do this all the time.

So you can use the same hardware for both a h=gateway and a LoRa Node.

Maarten

PS. I did not publuch the LoRa Node part yet. Especially since the ESP32 and ESP8266 devices consume more power and probably need an adapter tp function for a long time.

IoTThinks, 2020-05-15:
LMIC 6?
Do you mean LMIC from IBM or MCCI?

I'm trying https://github.com/matthijskooijman/arduino-lmic and https://github.com/mcci-catena/arduino-lmic

but not so much luck.
Which LoRaWAN do you think compatible with this ESP-1CH-Gateway?

Thanks a lot.

platenspeler, 2020-05-16:
Looking at the READ.ME file it is mcci catena.

platenspeler, 2020-05-16:
and LMIC 6. And compatibel is the wrong word. The hardware is compatible but the libraries of the single channel gateways and the sensor nodes are completely different. What are you trying to do?
Normally people try to use a different sensor node that consumes less power and has no wifi.

IoTThinks, 2020-05-16:
I use https://github.com/matthijskooijman/arduino-lmic
And it works with ESP-1CH-Gateway.

Thanks a lot.

closed 2020-05-16 by IoTThinks