things4u/ESP-1ch-Gateway #39: PlatformIO compile error

Luxtech, 2020-07-01:
I followed the steps as explained in this documentation:
https://things4u.github.io/Projects/SingleChannelGateway/HardwareGuide/3_Hardware_Install/31_PlatformIO.html

But when I compile the code in platformIO I get the following errors:

Compiling .pio/build/Gateway_138/libfc7/LoRaCode/LoRaCode.cpp.o
Archiving .pio/build/Gateway_138/lib0a5/libaes.a
lib/LoRaCode/LoRaCode.cpp:38:17: fatal error: ESP.h: No such file or directory

* Looking for ESP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP.h"
* Web  > https://platformio.org/lib/search?query=header:ESP.h
*

compilation terminated.
Archiving .pio/build/Gateway_138/lib1a9/libArduinoJson.a
*** [.pio/build/Gateway_138/libfc7/LoRaCode/LoRaCode.cpp.o] Error 1

I tried to search for a ESP.h library in platformIO but I was not able to find anything.

ChrSchultz, 2020-07-18:
Please post for which board you did built the program and post your platformio.ini content.
so we could help further.

Samuel-ZDM, 2020-07-30:
I have the same problem.

In the configGway.h file, #define _PIN_OUT 4.
Paltformio.ini is:

`[env:Gateway_21]
platform = espressif8266
board = d1_mini
board_build.mcu = esp8266
board_build.f_cpu = 80000000L
build_flags =
  -D _PIN_OUT=1
  -D _WIFIMANAGER=0
  -D _SPIFFS_FORMAT=0
  -D _OLED=0
  -D _DUSB=1
  -D _PROFILER=1
  -D _STAT_LOG=1
framework = arduino
upload_protocol = espota
board_build.flash_mode = qio
upload_speed = 115200
upload_port = 192.168.2.21

[env:Gateway_29]
platform = espressif8266
board = d1_mini
board_build.mcu = esp8266
board_build.f_cpu = 80000000L
build_flags =
  -D _PIN_OUT=2
  -D _WIFIMANAGER=0
  -D _SPIFFS_FORMAT=0
  -D _OLED=0
  -D _DUSB=1
  -D _PROFILER=1
  -D _STAT_LOG=0
framework = arduino
upload_protocol = espota
board_build.flash_mode = qio
upload_speed = 115200
upload_port = 192.168.2.29

[env:Gateway_72]
platform = espressif32
board = heltec_wifi_lora_32
framework = arduino
build_flags =
  -D _SPIFFS_FORMAT=0
  -D _WIFIMANAGER=0
  -D _OLED=0
  -D _DUSB=1
  -D _PROFILER=1
upload_protocol = espota
board_build.flash_mode = qio
upload_speed = 115200
upload_port = 192.168.2.72

[env:Gateway_138]
platform = espressif32
board = heltec_wifi_lora_32
framework = arduino
build_flags =
  -D _WIFIMANAGER=0
  -D _SPIFFS_FORMAT=0
  -D _OLED=1
  -D _DUSB=1
  -D _PROFILER=1
;  -D _JSONENCODE=1
;  -D _MAXSEEN=0
upload_protocol = espota
board_build.flash_mode = qio
upload_speed = 115200
upload_port = 192.168.2.138

; Nr. 54 has WIFIMANAGER sometimes set
; Also the Gateway is a T-Beam sensor for temperature and GPS (_GATEWAYNODE=1)
[env:Gateway_54]
platform = espressif32
board = heltec_wifi_lora_32
build_flags =
  -D _WIFIMANAGER=0
  -D _SPIFFS_FORMAT=0
  -D _DUSB=0
  -D _OLED=0
  -D _GATEWAYNODE=1
  -D _PROFILER=1
framework = arduino
upload_protocol = espota
board_build.flash_mode = qio
upload_speed = 115200
upload_port = 192.168.2.54
`

Luxtech, 2020-08-17:
> Please post for which board you did built the program and post your platformio.ini content.
> so we could help further.

I used the following platformio.ini buildoptions:

```
[env:Gateway_138]
platform = espressif32
board = heltec_wifi_lora_32
framework = arduino
build_flags = 
	-D _WIFIMANAGER=0
	-D _SPIFFS_FORMAT=0
	-D _OLED=1
	-D _DUSB=1
	-D _PROFILER=1
upload_protocol = espota
board_build.flash_mode = qio
upload_speed = 115200
upload_port = 192.168.2.138
```

acaciochinato, 2020-09-06:
Any updates? 

I was able to compile it with this problems:
identifier "WlanConnect" is undefined
identifier "msg_lLED" is undefined
identifier "resolveHost" is undefined
identifier "writeGwayCfg" is undefined
identifier "readSeen" is undefined
identifier "writeConfig" is undefined
identifier "WlanConnect" is undefined
identifier "printInt" is undefined
identifier "printInt" is undefined

But when I try to upload the code, I got this:

01:00:43 [DEBUG]: Options: {'esp_ip': '192.168.2.138', 'host_ip': '0.0.0.0', 'esp_port': 3232, 'host_port': 53758, 'auth': '', 'image': '.pio/build/Gateway_138/firmware.bin', 'spiffs': False, 'debug': True, 'progress': True, 'timeout': 10}
01:00:43 [INFO]: Starting on 0.0.0.0:53758
01:00:43 [INFO]: Upload size: 953264
Sending invitation to 192.168.2.138 ..........
01:02:23 [ERROR]: No response from the ESP
*** [upload] Error 1

Any ideas?

ChrSchultz, 2020-09-07:
Please write down your conditions ide etc. Did you use platformio or arduinoide.--Christoph Schultz Stütensen 229571 Rosche e-mail: ic-schultz@gmx.de

acaciochinato, 2020-09-07:
Yes. I'm using platformio. Here is my platformio.ini:

[env:Gateway_138]
platform = espressif32
board = heltec_wifi_lora_32_V2
framework = arduino
build_flags = 
	-D _WIFIMANAGER=1
	-D _SPIFFS_FORMAT=0
	-D _OLED=1
	-D _DUSB=1
	-D _PROFILER=1
upload_protocol = espota
board_build.flash_mode = qio
upload_speed = 115200
upload_port = 192.200.20.22

ChrSchultz, 2020-09-08:
So your command shoul be: 'platformio run -e Gateway_138 -t upload' . Try it out. So i had no problems with building.

acaciochinato, 2020-09-08:
Same error. The upload method should be via USB, right? Maybe it's trying to upload to the paramether upload_port ? I got the feeling that he's trying to connect and upload to that IP but is not getting any response... 

Is there any place that I need to specify the USB port?

Luxtech, 2020-09-08:
> So your command shoul be: 'platformio run -e Gateway_138 -t upload' . Try it out. So i had no problems with building.

Does not work, I tried compiling with platformIO in Visual studio code and with platformIO-core installed with PIP. Both give the same error:
```
$ platformio run -e Gateway_138 -t upload
Processing Gateway_138 (platform: espressif32; board: heltec_wifi_lora_32; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/heltec_wifi_lora_32.html
PLATFORM: Espressif 32 (1.12.4) > Heltec WiFi LoRa 32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - tool-openocd-esp32 1.1000.20190708 (10.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Converting ESP-sc-gway.ino
/home/lux/Downloads/ESP-1ch-Gateway-master/src/ESP-sc-gway.ino:108:0: warning: "ESP_getChipId" redefined
 # define ESP_getChipId()   (ESP.getChipId())
 ^
/home/lux/Downloads/ESP-1ch-Gateway-master/src/ESP-sc-gway.ino:84:0: note: this is the location of the previous definition
 # define ESP_getChipId()   ((uint32_t)ESP.getEfuseMac())
 ^
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <aes>
|-- <LoRaCode> 1.0
|-- <ArduinoJson> 6.14.0
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <SPI> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <Streaming> 5.0.0
|-- <Time> 1.6
|-- <TinyGPSPlus> 1.0.2
|-- <WebServer> 1.0
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <WiFi> 1.0
|-- <WifiManager> 1.0.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <WebServer> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <base64> 1.0.0
|-- <ESP8266_SSD1306> 3.2.7
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
Building in release mode
Compiling .pio/build/Gateway_138/src/ESP-sc-gway.ino.cpp.o
Compiling .pio/build/Gateway_138/libc65/LoRaCode/LoRaCode.cpp.o
lib/LoRaCode/LoRaCode.cpp:38:17: fatal error: ESP.h: No such file or directory

*************************************************************
* Looking for ESP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP.h"
* Web  > https://platformio.org/lib/search?query=header:ESP.h
*
*************************************************************

compilation terminated.
Compiling .pio/build/Gateway_138/lib2a9/WiFi/WiFiSTA.cpp.o
Compiling .pio/build/Gateway_138/lib2a9/WiFi/WiFiScan.cpp.o
Compiling .pio/build/Gateway_138/lib2a9/WiFi/WiFiServer.cpp.o
Compiling .pio/build/Gateway_138/lib2a9/WiFi/WiFiUdp.cpp.o
Compiling .pio/build/Gateway_138/libb02/ESPmDNS/ESPmDNS.cpp.o
*** [.pio/build/Gateway_138/libc65/LoRaCode/LoRaCode.cpp.o] Error 1
Compiling .pio/build/Gateway_138/libb1b/ArduinoOTA/ArduinoOTA.cpp.o
In file included from lib/ESP8266_Oled_Driver_for_SSD1306_display/SSD1306Wire.h:31:0,
                 from lib/ESP8266_Oled_Driver_for_SSD1306_display/SSD1306.h:30,
                 from src/oLED.h:60,
                 from /home/lux/Downloads/ESP-1ch-Gateway-master/src/ESP-sc-gway.ino:66:
lib/ESP8266_Oled_Driver_for_SSD1306_display/OLEDDisplay.h: In member function 'virtual bool OLEDDisplay::connect()':
lib/ESP8266_Oled_Driver_for_SSD1306_display/OLEDDisplay.h:253:29: warning: no return statement in function returning non-void [-Wreturn-type]
     virtual bool connect() {};
                             ^
/home/lux/Downloads/ESP-1ch-Gateway-master/src/_txRx.ino: In function 'int sendPacket(uint8_t*, uint8_t)':
/home/lux/Downloads/ESP-1ch-Gateway-master/src/_txRx.ino:100:11: warning: unused variable 'powe' [-Wunused-variable]
   uint8_t powe = root["txpk"]["powe"];   // power, e.g. 14 or 27
           ^
/home/lux/Downloads/ESP-1ch-Gateway-master/src/_txRx.ino:126:10: warning: unused variable 'w' [-Wunused-variable]
  int32_t w = (int32_t) (LoraDown.tmst - micros()); // Wait Time compute
          ^
========================================================================================================= [FAILED] Took 5.85 seconds=========================================================================================================

Environment    Status    Duration
-------------  --------  ------------
Gateway_138    FAILED    00:00:05.847
=================================================================================================== 1 failed, 0 succeeded in 00:00:05.847 ===================================================================================================

```

ChrSchultz, 2020-09-08:
Try inserting this in your platformio.ini :env:Gateway_140]platform = espressif32board = heltec_wifi_lora_32_V2framework = arduinobuild_flags =-D _WIFIMANAGER=1-D _SPIFFS_FORMAT=0-D _OLED=1-D _DUSB=1-D _PROFILER=1upload_protocol = esptoolboard_build.flash_mode = qioupload_speed = 115200upload_port  is Not used with esptool, cmd-line option  '-e Gateway_140' and connect device with usb. Upload_protocol  espota is for ota upload and you must specify upload_port with the correct IP address.

acaciochinato, 2020-09-08:
> Does not work, I tried compiling with platformIO in Visual studio code and with platformIO-core installed with PIP. Both give the same error:
>
> ```
> $ platformio run -e Gateway_138 -t upload [...]

You need to change "ESP.h" for "Esp.h" in LoRaCode.cpp

acaciochinato, 2020-09-08:
This. Thanks, that solved! Just changing the upload method!

platenspeler, 2021-07-11:
Thanks to all that helped!

closed 2021-07-11 by platenspeler