things4u/ESP-1ch-Gateway #40: REPEATER isn't working?

antonio-emilio, 2020-07-13:
I've been using this project for the last few days and it helped me alot. 
I was needing something to do like a "Mesh" network, using the ESP32 to receive a package and then send it again to the gateway.
This code has an "REPEATER" option, but It seens like it doesnt do nothing... am I correct?

platenspeler, 2020-07-16:
Yes you are correct.
 I am still working on this idea but since there was no interest I have not yet implemented  nor tested its functionality 100%.

Maarten

Verstuurd vanaf mijn iPad

antonio-emilio, 2020-07-16:
I got it. I made some adaptations, integrating another code with this one and it worked. 
I think it would be a good feature...

closed 2020-07-16 by antonio-emilio

AllTracking, 2022-07-07:
@antonio-emilio  Could you share the modified code?

@platenspeler thank you for such a library but why in the repeater mode it still requires the SSID(even when i set the _LOCALSERVER=0)? i would like to test the Repeater functionality in a place where no wifi signal or ethernet exists.