things4u/ESP-1ch-Gateway #41: can I "fake" the connection to the internet?

antonio-emilio, 2020-07-26:
I've doing some modifications to this code and I added the functionality to re-send the received package (Repeater Mode). But this would only make sense if I could take out the necessity of being connected to the internet. Is that possible?

Samuel-ZDM, 2020-07-30:
Hello!

I am trying to do that too.
I want to receive a message from a node and repeat it to another gateway.
I haven't been able to do that for now.

If anyone can help us.

antonio-emilio, 2020-07-30:
> Hello!
> 
> I am trying to do that too.
> I want to receive a message from a node and repeat it to another gateway.
> I haven't been able to do that for now.
> 
> If anyone can help us.

I included some code of another library and made it work, but I still need to take of the necessity of being connected to the internet (otherwise it will not make sense).
The code I used was this one (https://github.com/matthijskooijman/arduino-lmic)...you just have to make the logic to send the content you just received.
(BR rules... em ingles porque vai que ajuda mais alguem rs)

Samuel-ZDM, 2020-07-30:
Yes. If it is necessary for the devices to be connected to the Internet, it makes no sense.

I am also using this LMIC library on my nodes. So, I have to edit the gateway code with the LMIC functions to be able to forward the packets. I will try to do that.

(Yes, continuamos em inglês kkk)
Thanks for the reply.