things4u/ESP-1ch-Gateway #46: a lot of uplinks missed

antonio-emilio, 2020-08-11:
First of all, I want to thank the creators of this repository. Such a useful code!
I've been missing a lot of uplinks. Last week I made my first test outside, but i received a lot of messages like this one:
"Thu 11-08-2020 17:04:42- SCAN:: I=CDDONE CDDETD , F=0, SF=7, E=0, S=RX , eT=13414520, dT=2241""
instead of the  UP RXPK.

It works, but I've been missing a lot of packages... is there something I can do?

hujiese, 2020-08-12:
I met the same problem, set _CAN=0 may work bette,but the problem is still occur.

antonio-emilio, 2020-09-04:
When I disabled CAD, it worked better, but I still have a lot of problems.. For example, if I try to send two packages with a tiny delay, I only get the first package.... Seems like the gateway only recognize packages with some delay between them....

platenspeler, 2020-09-06:
I cannot find more examples right now. But you can see two sensor receptions within a few seconds.
I have not experienced delays or so, but I will pay more attention next days.

Sun 06-09-2020 23:21:32               lora-43 temp DS18B20    0              868100000           7              -66
Sun 06-09-2020 23:21:28               lora-40 airquality              0              868100000           9              -85

Regards,

Maarten

Samuel-ZDM, 2020-10-22:
I had the same problem. I manage to fix by updating the code of the final device. I put only one frequency for sending data and it reduces packet loss.

antonio-emilio, 2020-10-29:
> I met the same problem, set _CAN=0 may work bette,but the problem is still occur.

It was the best option for me.... not the solution, but at least worked better.

closed 2020-10-29 by antonio-emilio