things4u/ESP-1ch-Gateway #51: Gateway does not send downlink to APB device

Samuel-ZDM, 2020-10-13:
Hello, I'm trying to send a downlink message to APB devices.
I configured my gateway for US902_928.
The downlink to TTN must be SF9, but the terminal shows SF12. The file configGway.h has _RX2_SF 9, as it should be.

Terminal:

sendPacket:: token=1536, {"txpk":{"imme":false,"tmst":407270414,"freq":923.3,"rfch":0,"powe":20,"modu":"LORA","datr":"SF12BW500","codr":"4/5","ipol":true,"size":22,"ncrc":true,"data":"YMgXAyaKBQADQAIAcQM1AP8BZUMRKg=="}}

txLoraModem:: end=micr=309.606519, tmst=309.606678, wait=159, freq=923.299, **sf=12, bw=244**, powe=15, crc=0, imme=0, iiq=40, prea=8, rfch=0, ncrc=1, size=22, strict=1, a=26:03:17:c8:, size=22, data

NicoThien, 2020-10-27:
if  "_STRICT_1CH" = 0 ?

platenspeler, 2020-11-16:
Hi Samuel,

I see that you are right. The software (even very close to the hardware) says that the downlink is complete and I even receive an interrupt TXDONE. However I see that not all is transmitted.

The latest version 6.2.7 does include separate uplink and downlink monitoring/website. If you see what;s going wrong please tell me.

Maarten

Samuel-ZDM, 2020-11-16:
Thank you.

I'll do the tests, when I finish I give you feedback.

Samuel-ZDM, 2020-11-26:
Now it's working. But I found some problems.

One problem I found now is in the downlink message counter that is not counting.
[image]

Another problem is that it is no longer printing the sending outputs, with the channel and SF used, as in the question I asked that I had all the information.
[image]

I will try to help fix these errors if possible.

Thank you!

Samuel-ZDM, 2020-11-26:
It's still not working.
I took a test now and no messages are received. Even if the SF problem is different. 
When I look for this variable _RX2_SF, I don't find it being used anywhere. Perhaps the value of the SF downlink is not being changed.

Thank you!

ncamilo, 2020-12-14:
In my case the Downlink frequency is wrong, I'm searching where to correct it, anyone found another issue to this?

Thanks!

cbijon, 2021-02-15:
Hello,

Do you have some news about this issue ? how you resolv / workarround it ?

Best,

Charles

itsygithub, 2021-04-05:
I have corrected frequency in json file which will also solve the problem with downlink. I created a pull request from my fork.

ncamilo, 2021-07-09:
I correct reconfiguring the gateway for my region because in Brazil we use AU915 instead of US915.

Best, thanks for all!