things4u/ESP-1ch-Gateway #52: OLED does not working

lordrak007, 2020-10-16:
Hello,
i have Heltec Wireless Stick with oled.
I have enabled oled in setting. I have check pinout setting with heltec library and all looks fine. Unfortunatelly oled does nothing.
Have you any idea why or how to debug?
Here is serial debug:
SPIFFS begin
Do Asserts
.CH=    0
.SF=    9
.FCNT=  0
.DEBUG= 1
.PDEBUG=        96
.CAD=   1
.HOP=   0
.NODE=  0
.BOOTS= 8
.RESETS=        0
.WIFIS= 6
.VIEWS= 0
.REFR=  0
.REENTS=        0
.NTPETIM=       0
.NTPERR=        0
.WAITERR=       0
.WAITOK=        0
.NTPS=  6
.FILEREC=       0
.FILENO=        0
.FORMAT=        0
.DELAY= 0
.TRUSTED=       1
.EXPERT=        0
.SEEN=  1
.MONITOR=       1
readGwayCfg:: return OK
MAC: 10:52:1c:5a:d1:30, len=17
0:0:2. WlanConnect SSID=uLIOT
WlanStatus:: CONNECTED ssid=uLIOT
Host=esp32-5ad130 WiFi Connected to uLIOT on IP=192.168.1.46
Local UDP port=1700
UDP Connection successful
Gateway ID: 10521cffff5ad130, Listening at SF9 on 868.10 MHz.
resolveHost:: OK=nl.pool.ntp.org IP=162.159.200.123
Time set=Fri 16-10-2020 09:53:16
resolveHost:: OK=router.eu.thethings.network IP=52.169.76.203
setupOta:: Started
Ready IP address: 192.168.1.46
WWW Server started on port 80
OLED_ADDR=0x3C
 --- Setup() ended, Starting loop() ---

platenspeler, 2020-11-16:
I cannot see whats going wrong here. The OLED_ADDR setting is not really a debug message, but it does support the known OLED displays. Pin setting is 4 for Heltec.

What version of Heltec board are you using?

Thanks,

Maarten

CamFlyerCH, 2020-12-27:
I use a TTG0 Lora32 and OLED is only working with  "#	define _PIN_OUT 5" in configGway.h (and ESP-SC-gway.ino) instead of 4