things4u/ESP-1ch-Gateway #53: Gateway Reboots after Downlink attempt

kulsoomjafri, 2020-10-29:
Hi,
Sincere compliments and kudos for the good work. I have been using gateway since version "V.5.3.3.H; 180825a", now upgraded at "V.6.2.3.E.EU868; PlatformIO 200223a" 
The issue arose in V.5.3.3 whenever I sent a message. I realized that payLength was zero. I was running into the issue on V.5.3.3 which I fixed locally by adding the following in _loraModem.ino line 780
```
 if (payLength>0)
 {
	sendPkt(payLoad, payLength);
 }
```

The error in this version is as follows:

Stack smashing protect failure!

abort() was called at PC 0x400f3960 on core 1

Backtrace: 0x4008cc2c:0x3ffb18c0 0x4008ce5d:0x3ffb18e0 0x400f3960:0x3ffb1900 0x400d6cde:0x3ffb1920 0x400d71e5:0x3ffb1980 0x400e042a:0x3ffb19f0 0x400e2ca3:0x3ffb1f60 0x400ef4c5:0x3ffb1fb0 0x40089341:0x3ffb1fd0

That translates to :
Decoding stack results
0x4008cc2c: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008ce5d: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x400f3964: __stack_chk_fail at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/stack_check.c line 36
0x400d6cde: printDwn(LoraDown*, String&) at C:\Users\KJ\Documents\Arduino\ESP-1ch-Gateway-master_v6\ESP-sc-gway/_utils.ino line 94
0x400d71e9: txLoraModem(LoraDown*) at .../_loraModem.ino line 801
0x400e042e: readUdp(int) at .../_udpSemtech.ino line 342
0x400e2ca7: loop() at .../ESP-sc-gway.ino line 690
0x400ef4c9: loopTask(void*) at ...\main.cpp line 19
0x40089341: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

I tried the same approach and tried fixing it as 
`  if (LoraDown->size>0)
`
But I still see the reboot issue. For now reverting back to V5.3.3

platenspeler, 2020-10-30:
Hi,

Too bad that you have troubles with the code. 
At this moment I am working on release 6.2.7.
Would it be possible to try the latest version, which would enable me to repair obvious errors?

Thanks,

Maarten

kulsoomjafri, 2020-10-30:
Thanks a lot for your response. Yes I will be able to test the latest
version. Thanks again for all your efforts to the community.

platenspeler, 2020-11-16:
The patest version on the site is 6.2.7 and has lots of issues fixed. I advise you to download the version.

I am now looking into the downlink issue. It seems that the software expects all downlinks to happen, but not all of them are actually sent.

kulsoomjafri, 2020-11-23:
Thank again Maarten for reaching out to me. I have downloaded the latest
version but still having the same issue of downlink messages. I also
corrected some other things in the code which I think will be useful for
the users who experience restarting of their gateways after every downlink
message sent from the TTN. For your reference, I am attaching the
screenshot. It is located in loraModem.io
[image: image.png]
[image: image.png]