things4u/ESP-1ch-Gateway #56: Problem joining OTAA / Downlink in general

Sprattler, 2020-11-18:
After i probably tried everything. I tried V5 with the V5 with the Dowlink Fix which lead to the same Problem.
I am able to connect to my Gateway which gives an "Join Accept". but obviously it doesnt.

In ABP Uplink works (unstable) but Downlink doesn't.

[image]
(Gateway Traffic in TTN)
---------------------------------------------------------------------------------------------------------------------------------------------
[image]
(Device Traffic)
---------------------------------------------------------------------------------------------------------------------------------------------
[image]
Arduino DebugSerial of the Example OTAA Join 
---------------------------------------------------------------------------------------------------------------------------------------------

Thanks Mates,
greetings Leon

platenspeler, 2020-11-19:
I know that downlink does not work for some reason. There are too many gateways in the neighbourhood too test easy myself, but I was told some time ago that it worked.
I will test and try again. Please find my latest code in the GIT which tells me rhat it sends the data and even issues a TXDONE. But obviously it has difficulties.

Maarten

ArnieO, 2021-03-20:
Which is the latest version where OTAA is known to work? 
I am struggling with this, and I am so "lucky" there are no other gateways in the vicinity that "pollutes" my testing.

IoTThinks, 2021-04-14:
Last time, my downlink is working with ABP.
Wonder anyone is able to make OTAA to work?

Thanks a lot.