things4u/ESP-1ch-Gateway #57: Heltec WiFi LoRa 32 V2

bibibo-77, 2020-11-19:
Hello,

i try to compile this app by Ardoino IDE but i still get some errors. I am not sure where is problem. Please could you help me ?

ESP-sc-gway:403:22: error: 'WlanConnect' was not declared in this scope
  while (WlanConnect(0) <= 0) {
                      ^
ESP-sc-gway:508:25: error: 'msg_lLED' was not declared in this scope
  msg_lLED("GET TIME",".");
                         ^
ESP-sc-gway:509:44: error: 'resolveHost' was not declared in this scope
  ntpServer = resolveHost(NTP_TIMESERVER, 15);
                                            ^
ESP-sc-gway:559:41: error: 'writeGwayCfg' was not declared in this scope
   writeGwayCfg(_CONFIGFILE, &gwayConfig );
                                         ^
ESP-sc-gway:594:30: error: 'readSeen' was not declared in this scope
  readSeen(_SEENFILE, listSeen);       // read the seenFile records
                              ^
ESP-sc-gway:631:38: error: 'writeConfig' was not declared in this scope
  writeConfig(_CONFIGFILE, &gwayConfig);     // Write config
                                      ^
/Users/peter/Documents/Arduino/ESP-sc-gway/ESP-sc-gway.ino: In function 'void loop()':
ESP-sc-gway:671:19: error: 'WlanConnect' was not declared in this scope
  if (WlanConnect(1) < 0) {
                   ^
ESP-sc-gway:844:31: error: 'printInt' was not declared in this scope
    printInt(micros(), response);
                               ^
ESP-sc-gway:864:31: error: 'printInt' was not declared in this scope
    printInt(micros(), response);

bibibo-77, 2020-11-19:
It is solved.

fedyfausto, 2021-02-05:
how?

slater0013, 2021-02-13:
Same compile errors here ! @bibibo-77 would be great if you share with us how you sorted this out !
Thanks,
GS.

slater0013, 2021-02-13:
Ok my Bad ! I did not read the README completely :

Arduino IDE
Create a place on you filesystem to work on the files. In this directory create the source directory "ESP-sc-gway" and the libraries directory "libraries". When unpacking the source at github: Copy the content of the "src" directory to the Aruino IDE "ESP-sc-gway" directory and copy the contents of the "lib" directory to the Arduino IDE "libraries" directory;

In case someone finds this post through a google search