things4u/ESP-1ch-Gateway #63: MQTT Documentation missing

bdemers, 2020-12-28:
There are a few references to MQTT throughout the docs and code.  I wasn't able to find the code that is actually sending any MQTT messages.  

Specifically, I'm trying to figure out how to route data from my nodes (sensor data) to a private MQTT server, with a specific topic.

Is this possible or am I misunderstanding the comments/doc references to MQTT?

hisnameishb, 2021-02-14:
You subscribe to the mqtt server. What you receive from there you can send to your own mqtt.

bdemers, 2021-02-14:
> You subscribe to the mqtt server from TTN. What you receive from there you can send to your own mqtt.

Right 😉 
But documentation (and an example) around this would be great.