things4u/ESP-1ch-Gateway #77: ESP.h file required for compilation but not included in lib folder .....

olive2205, 2021-04-19:
all is in the title ...

amauryverschooren, 2021-04-23:
If you select an ESP board in the IDE the error wil go away.

takurx, 2021-06-02:
I also got compile error on Ubuntu 20.04 and VScode 1.56.2, Platformio Core 5.1.1 and Home 3.3.4
```
lib/LoRaCode/LoRaCode.cpp:38:17: fatal error: ESP.h: No such file or directory
```

I had install platforms that is Espressif32 and Espressif 8266.
I can success to compile on windows 10, VScode and Plafformio. It's strange......

PSF1, 2023-06-11:
In [esp8266.com](https://www.esp8266.com/viewtopic.php?f=33&t=3123#p17763), [martinayotte](https://www.esp8266.com/memberlist.php?mode=viewprofile&u=138) say: _it is "Esp.h" (case sensitive on Linux)_