things4u/ESP-1ch-Gateway #82: Compile fails, reference to 'byte' is ambiguous

gjt211, 2021-06-02:
Hi, downloaded zip as per instruction, copied files & libraries as directed.
Followed steps in "Getting Started" section, only edited wifi credentials.
Click compile and get lots of errors. First error is
```
ESP-sc-gway:146:1: error: reference to 'byte' is ambiguous
  146 | byte currentMode = 0x81;
      | ^~~~
```
These continue and with variations.
I am using a generic ESP8266 with RFM95, but haven't even got to program it yet.

nigelorr, 2021-06-21:
What are you using to build (version number of Arduino / PlatformIO), have you successfully built other code for the same board on that setup?

If not, try that first, it looks like the build tool isn't set up correctly for the board you are targeting.

nablabla, 2021-07-05:
Hi, same problem here, the actual version for wemos is 3.0.1, which version should we use? the helltec esp8266 libraries in the wikies are not to be found :(

nablabla, 2021-07-09:
okay, i got PlatformIO installed. But it has the same problem. As far as I understood it, it should work out of the box in PlatformIO by pulling its dependencies, but then - this happened:
C:/Users/<user>/Documents/PlatformIO/Projects/ESP-1ch-Gateway/src/_utils.ino:395:2: error: reference to 'byte' is ambiguous
  395 |  byte _day = day(eTime);

dev-strom, 2021-07-09:
It seems to be a bug in the Espressif framework since version 3.0.0
See https://github.com/esp8266/Arduino/issues/8089

As a workaround you can use an older version of the framework.
Change in your `platformio.ini` the line
```
platform = espressif8266
```
to
```
platform = espressif8266@2.6.3
```

d-a-v, 2021-07-10:
esp8266/Arduino v3+ now uses gcc10's c++17 which defines the `std::byte`.
As a result, using Arduino's `byte` with `using namespace std;` is incompatible. A temporary fix could be to locally remove `using namespace std;` and use `std::stuff` instead of `stuff` where relevant.
Removing Arduino's `byte` in favour of `using std::byte;` is not an option because `byte test = 0` is an error (`int` incompatible with `byte`...)
The esp8266/Arduino core team might come with a 3.0.2 release that will apply a dirty hack to remove the stdc++17 `byte` definition.

In the meantime, release 2.7.4 is more recent that 2.6.3 which still uses gcc4.8 (maybe #62 is needed, I wonder why there's no feedback)
```
platform = espressif8266@2.7.4
```

nablabla, 2021-07-10:
Error: Could not find the package with 'espressif8266 @ 2.7.4' requirements for your system 'windows_amd64'
it appears to not exist. But I am not shure, I just switched to PlatformIO, years ago I was using MS visual micro but the community seems to go the platformIO path instead. It feels so crappy to me. I get a lot of random error messages which I can not make sense of :(

d-a-v, 2021-07-10:
My bad.
PlatformIO has a different version scheme. [2.6.3 is the right version number](https://github.com/platformio/platform-espressif8266/releases).
There's an [ongoing proposal](https://github.com/esp8266/Arduino/pull/8208) to fix this issue in the esp8266 arduino core v3.x.
(but #62 is still needed I think).

nablabla, 2021-07-10:
hey thank you for your help, i got it working with espressif8266@2.6.3
but wouldn't it be better if we add the "@x.x.x" to _every_ dependency in platformIO projects? so we can benefit with stability and guarantee that it works in the future

d-a-v, 2021-07-10:
That's a question to ask to platformIO