things4u/ESP-1ch-Gateway #84: Dual Channel LoRaWAN gateway?

IoTThinks, 2021-06-26:
Is it possible to extend the library to support Dual Channel LoRaWAN gateway?
Inspired from https://forum.chirpstack.io/t/dragino-lg01-n-no-live-lorawan-frames/11377/6

If we have an ESP32 board with 2 SX127x modules and we can support 2 channels, then we could use most of the LoRaWAN functionalities.

Most LoRaWAN frequency plans only require a minimum of 2 frequencies for OTAA....

Zupayruna1, 2022-05-25:
I'm working in something similar, maybe we can help each other. I'm intested in add a new channel in the lora-gateway-esp32, but I'm not sure if this is more easy add channel for uplink or downlink?

IoTThinks, 2022-06-19:
Stuck with daily work.
May be easier to try with two single gateway instead?
Each gateway to listen for a frequency.

Zupayruna1, 2022-07-13:
I have the first example for a gateway with two channel for uplink based on ESP32

IoTThinks, 2022-09-13:
> I have the first example for a gateway with two channel for uplink based on ESP32

Would you mind sharing it?
I will be back to this issue soon end of Sep.

Thanks millions.

Zupayruna1, 2022-09-13:
> > I have the first example for a gateway with two channel for uplink based on ESP32
> 
> Would you mind sharing it? I will be back to this issue soon end of Sep.
> 
> Thanks millions.

Hello 
Sure, I was uploading the files, maybe you can check the code. For until now the gateway work without system errors for long time. https://github.com/Zupayruna1/update_LoraWan_gateway_ESP32
Best regards!

IoTThinks, 2022-09-18:
Great. Thanks a lot.
Let me try.