things4u/ESP-1ch-Gateway #87: TTGO  Lora OLED doesn't show anything

AnakinSpain, 2021-07-24:
With my TTGO 868/915mhz Model: LORA32 screen is off.

I change pinout to 4 an oled to 1, but it doesn't work.

orimate, 2021-07-26:
For which environment "[env:Gateway_38"] did you compile? And exactly which board do you have? You could give a little bit more information.

matthew73210, 2022-10-15:
Hey, any luck with this?
What ever I set the pin out to, the oled won't work, and i'f I change it in oled.h the thing won't compile at all
Cheers

ChristianRoze80, 2023-12-07:
TTGO T3_V1.6.1 doesnt use "OLED_RST"  Pin.  You can edit "_oLED.h"  and comment it out.

#elif _PIN_OUT==5								// TTGO with external OLED
#define OLED_SCL 22								// GPIO22 / SCL
#define OLED_SDA 21								// GPIO21 / SDA
//#define OLED_RST 16								// Reset pin (Some OLED displays do not have it)