things4u/ESP-1ch-Gateway #89: no data on aplication

navoraul, 2021-08-02:
Hello, I send data from a RAK4600 , and I see in the console activity ok, but I do not see nothing in the aplication-device live data section.

I am using ABP activation in TTN V3

navoraul, 2021-08-02:
{
  "name": "gs.up.receive",
  "time": "2021-08-02T21:53:57.224297839Z",
  "identifiers": [
    {
      "gateway_ids": {
        "gateway_id": "gatewayeuropeo"
      }
    },
    {
      "gateway_ids": {
        "gateway_id": "gatewayeuropeo",
        "eui": "3C6105FFFF0D37B8"
      }
    }
  ],
  "data": {
    "@type": "type.googleapis.com/ttn.lorawan.v3.UplinkMessage",
    "raw_payload": "QJY6CyaAAQAIU19Iu5k=",
    "payload": {
      "m_hdr": {
        "m_type": "UNCONFIRMED_UP"
      },
      "mic": "X0i7mQ==",
      "mac_payload": {
        "f_hdr": {
          "dev_addr": "260B3A96",
          "f_ctrl": {
            "adr": true
          },
          "f_cnt": 1
        },
        "f_port": 8,
        "frm_payload": "Uw=="
      }
    },
    "settings": {
      "data_rate": {
        "lora": {
          "bandwidth": 125000,
          "spreading_factor": 7
        }
      },
      "coding_rate": "4/5",
      "frequency": "868099975",
      "timestamp": 3045960177
    },
    "rx_metadata": [
      {
        "gateway_ids": {
          "gateway_id": "gatewayeuropeo",
          "eui": "3C6105FFFF0D37B8"
        },
        "timestamp": 3045960177,
        "rssi": -57,
        "channel_rssi": -57,
        "snr": 10,
        "uplink_token": "ChwKGgoOZ2F0ZXdheWV1cm9wZW8SCDxhBf//DTe4EPHTtqwLGgsI9dKhiAYQtpHvaiDoqv+K01g="
      }
    ],
    "received_at": "2021-08-02T21:53:57.224118966Z",
    "correlation_ids": [
      "gs:conn:01FC4AXPJZNQEAY80W1T7ZMDQE",
      "gs:uplink:01FC4DQWF829PT3MXPH0368XYP"
    ]
  },
  "correlation_ids": [
    "gs:conn:01FC4AXPJZNQEAY80W1T7ZMDQE",
    "gs:uplink:01FC4DQWF829PT3MXPH0368XYP"
  ],
  "origin": "ip-10-100-12-248.eu-west-1.compute.internal",
  "context": {
    "tenant-id": "CgN0dG4="
  },
  "visibility": {
    "rights": [
      "RIGHT_GATEWAY_TRAFFIC_READ",
      "RIGHT_GATEWAY_TRAFFIC_READ"
    ]
  },
  "unique_id": "01FC4DQWF8KQ21RX0H3YTZXAYD"
}

cstratton, 2021-10-02:
TTN explicitly prohibits attempting to connect single and dual channel fake gateways such as this to their servers

Gilverton, 2022-02-28:
Please look at this solved issue [Link](https://github.com/things4u/ESP-1ch-Gateway/issues/101) 
It is similar to what happened to you and did the trick in my case. ;)

cstratton, 2022-02-28:
You are mistaken; there can be no software solution to the actual problem, because the hardware is simply not capable of doing the job of a gateway.  LoRaWAN requires that a gateway simultaneously monitor 42 frequency and spreading factor combinations, but this device can only monitor one.  When you try to use this, you create a misleading situation where a gateway exists for only one combination, but does not exist for any of the other combinations that a proper LoRaWAN node is not just allowed, but actually _required_ to make use of.

This completely screws up the ADR algorithm for settings spreading factor, and the fair channel selection algorithm required not only by the LoRaWAN spec, but actually _by law_.

Worse, when you do this on a public network, your mess things up for other people's nodes, even if you're not noticing the problem on you own nodes that you may have rigged to violate the LoRaWAN spec and radio laws, but only using a single frequency and spreading factor.