things4u/ESP-1ch-Gateway #90: Error if compiling as Node

NL647, 2021-08-02:
Hello , i compiled the code as Gateway and if works fine.But if i add the option for node (_GATEWAYNODE=1) I get an error while compiling.
Do someone know how to fix it?
i'm using Lora heltech v2 and platfformio .

```
build_flags =
  -D _WIFIMANAGER=0
  -D _SPIFFS_FORMAT=0
  -D _OLED=1
  -D _DUSB=1
  -D _GATEWAYNODE=1
  -D _PROFILER=1
  -D _STRICT_1CH=2
  -D _MAXSEEN=10
  -D _STAT_LOG=0
  -D _LOCALSERVER=2
  -D _BUF_WRITE=0
  -D _EXPERT=1
```

```
`Compiling .pio\build\Gateway_38\lib326\WiFi\WiFiMulti.cpp.o`

`C:/Users/INFORMATIQUE/Desktop/ESP-1ch-Gateway-master/src/ESP-sc-gway.ino: At global scope:
C:/Users/INFORMATIQUE/Desktop/ESP-1ch-Gateway-master/src/ESP-sc-gway.ino:203:17: error: expected initializer before '.' token`
 ` uint16_t LoraUp.fcnt = 0;        // We write this to SPIFF file`
                 ^
`C:/Users/INFORMATIQUE/Desktop/ESP-1ch-Gateway-master/src/ESP-sc-gway.ino:204:19: error: expected initializer before '.' token
  uint16_t LoraDown.fcnt = 0;        // LoraDown framecount init 0`
                   ^
`C:/Users/INFORMATIQUE/Desktop/ESP-1ch-Gateway-master/src/_loraFiles.ino: In function 'int readGwayCfg(const char*, espGwayConfig*)':
C:/Users/INFORMATIQUE/Desktop/ESP-1ch-Gateway-master/src/_loraFiles.ino:137:18: error: 'struct espGwayConfig' has no member named 'fcnt'
   if (gwayConfig.fcnt != (uint8_t) 0) {`
                  ^
`C:/Users/INFORMATIQUE/Desktop/ESP-1ch-Gateway-master/src/_loraFiles.ino:138:29: error: 'struct espGwayConfig' has no member named **'fcnt'**`
    `LoraUp.fcnt = gwayConfig.fcnt+10;   // Assume it is only 10 off`

```
Thank for helping.

steengron, 2021-08-16:
Hi, I have the same problem. Have you found a solution?

ABurduli, 2022-02-06:
Have you found a solution?

steneor, 2022-02-06:
comment lines 202 to 205
202 //#if _GATEWAYNODE==1
203 //	uint16_t LoraUp.fcnt = 0;								// We write this to SPIFF file
204 //	uint16_t LoraDown.fcnt = 0;								// LoraDown framecount init 0
205 //#endif