things4u/ESP-1ch-Gateway #93: Error sending data to TTN V3

Samuel-ZDM, 2021-08-11:
Hello!

I'm managing to send the data to the gateway, but when the gateway sends it to TTN V3 it's giving an error and consequently it doesn't update the information in the console.

My settings are:

> // TTN Version 3:
> # define _TTNSERVER "nam1.cloud.thethings.network"
> # define _TTNPORT 1700 // Standard port for TTN

On the serial it looks like this:

> 4:41:51.004 -> wwwServer.ino:: ERROR upDown
> 14:41:51.004 -> wwwServer.ino:: ERROR upDown
> 14:41:51.004 -> wwwServer.ino:: ERROR upDown
> 14:41:58.204 -> ^ receivePkt:: rxPkt: t=Wed 08-11-2021 14:41:57, f=0, sf=7, a=26:0c:d1:e:, flags=50, addr=0, len=16, ERROR No Index in inDecodes

RCB07, 2021-08-14:
I have the same problem :(

Samuel-ZDM, 2021-08-15:
I changed the port value to 8887, the error "ERROR upDown" doesn't appear again, but it still doesn't work.
The error "Index in inDecodes" keeps appearing.

I'm using it in Brazil, TTN v2 had a specific TTN server, now I believe it doesn't. I don't know what the problem could be.

cstratton, 2021-10-02:
TTN vs explicitly prohibits connecting this device, so you need to stop even trying to do so.