things4u/ESP-1ch-Gateway #94: End of the one channel gateways ?

florafrisia, 2021-08-20:
Hi,

I'm searching for the solution why my gateway is not working.
I came across this article from the things industries
https://www.thethingsnetwork.org/forum/t/single-channel-packet-forwarders-scpf-are-deprecated-and-not-supported/31117

Can someone elaborate a little more about this matter that scpf are not supported anymore.

With kind regards,

Peter

RCB07, 2021-08-20:
Of course dont work :( Its very sad

shybzzz, 2021-08-30:
I can see this article is of 2019, and it worked somehow till 2021 :)

but current version of master branch of this repository does not work form me with ttn v3

cstratton, 2021-10-02:
Weather or not is appears to work to even the slightest extent is irrelevant, even _trying_ to connect these to TTN is explicitly prohibited.

JamesOlvertone, 2021-12-08:
I am not 100% in the LoRaWAN spec but would it be possible to mimic to TTN-Server 100% compliant Nodes while in the backend sensor nodes could be anything, even raw Lora or anything else as long as you have the right receivermodule for it. I mean the whole back and forth communication from TTN through the gateway to the node should end in the gateway. If TTN sends some requests from the sensor the gateway should send the data TTN expects. The gateway manages my local nodes with keys, IDs, Join and anything else what TTN uses. This is mapped to my local sensors that can be anything.

cstratton, 2021-12-23:
>would it be possible to mimic to TTN-Server 100% compliant Nodes while in the backend sensor nodes could be anything

Possibility would be irrelevant - doing so is a violation of TTN's terms of use.  

TTN is a public **LoRaWAN** network, if you want to do something that is not LoRaWAN, you have to run your own servers to handle your unique traffic.  The public server instances exist only to support proper LoRaWAN traffic from the user community.