things4u/ESP-1ch-Gateway #104: Error Unknown transceiver=0 Reboot

seb201, 2021-12-20:
Hello,

i have a Heltec Wireless Stick Lite (Panda).
For this board i have selected PIN_OUT==5 for ESP32/Heltec Wifi in the configGway.h, which should have the same pins as my board. Unfortunately i always get this error message in the console and the board restarts:

Unknown transceiver=0, pins.rst =14, pins.ss  =18, pins.dio0 =26, pins.dio1 =35, pins.dio2 =34
initLoraModem, unknown transceiver?
initLoraModem, unknown transceiver?
abort() was called at PC 0x400e0134 on core 1

Wireless Stick Lite Pinout Diagram:
[(https://resource.heltec.cn/download/Wireless_Stick_Lite/Wireless_Stick_Lite.pdf)]

Maybe someone can help me. Thanks

Willh-AM, 2022-03-18:
I had this same issue but with esp32 dev kit and a SX1276 LoRa module.

Zupayruna1, 2022-08-20:
> Hello,
>
> i have a Heltec Wireless Stick Lite (Panda). For this board i have selected PIN_OUT==5 for ESP32/Heltec Wifi in the configGway.h, which should have the same pins as my board. Unfortunately i [...]

I have the same board but I had to change the default pins in the code there is an error for wireless stick lite.