things4u/ESP-1ch-Gateway #105: Query: Integration with Chirpstack

oshin94, 2021-12-22:
Does this single channel gateway integrate with Chirpstack?
if Yes, please suggest which documentation to refer to.
if No, please suggest if there is any alternate way to integrate it.

G-3r4-d, 2021-12-28:
I tried it. It did not work. 

Unfortunately, I don't know any alternatives besides buying a cheap LoRaWAN gateway like TTIG. I avoided more headatches. 

I hope you can find a solution for your case

den3y, 2022-01-04:
I am currently trying it with Chirpstack in Kubernetes.
After configuring the chirpstack gateway bridge to have port 1700 open I can see the 1CH Gateway communicates with Chirpstack (also the "Last seen at" is updated every minute so I suppose it works).
I am using the Sparkfun ESP32 LoRa 1-CH Gateway.

Nevertheless I was not able to transmit data from my node via the 1CH Gateway to Chirpstack, this is what I am currently struggling with.

For the Gateway you simply need to set the variable _TTNSERVER to point to the IP or DNS-name of the server where the Chirpstack gateway bridge is located. You also need to set the corresponding _TTNPORT. Don't forget that the communication is UDP.

Westiewombat, 2022-04-30:
I do have this working + integrated with InnoDB - give me a few days to collate my config - yes single channel will **only** work with ABP

Isakvdw, 2022-07-16:
> I do have this working + integrated with InnoDB - give me a few days to collate my config - yes single channel will **only** work with ABP

Can you elaborate on why it would only work with ABP, and is that restriction specific to Chirpstack or does it apply to other platforms as well? I've been struggling to get my sensors working with this and chirpstack for the past few days, will try APB now but curious as to why OTAA won't work, thanks.

IoTThinks, 2022-08-14:
> > I do have this working + integrated with InnoDB - give me a few days to collate my config - yes single channel will **only** work with ABP
> 
> Can you elaborate on why it would only work with ABP, and is that restriction specific to Chirpstack or does it apply to other platforms as well? I've been struggling to get my sensors working with this and chirpstack for the past few days, will try APB now but curious as to why OTAA won't work, thanks. Im on eu868 freq

OTAA requires at least 2 mandatory channels for the gateway.
So single channel gateway can not be used for OTAA.

IoTThinks, 2022-08-14:
I try this fix https://github.com/things4u/ESP-1ch-Gateway/pull/72?fbclid=IwAR3NNYZeltljZ69OGcGXJc8ND5zKeoKk2JFBDRrtW7CL01TNnDqvuqLl6ZY#issuecomment-1214316043

And can integrate with ChirpStack + ABP.
[image]

Isakvdw, 2022-08-14:
Thanks, I temporarily got around it by editing the library but will try that as it does look like a better solution.

Zupayruna1, 2022-08-20:
> I'm trying to make connections to ChirpStack, but the gateway doesn't show up and no messages from ABP devices reach the ChirpStack server.
> 
> I put the values in _THINGSERVER and _THINGPORT

Yes only u need change these values in the gateway, and in more up in the same sketch there is another variable for TTNPort, in the gateway nothing more. If u continue with problems the problem is the server not 1ch gateway

Westiewombat, 2022-10-11:
The single channel gateway is just that. Lorawan uses multiple channels for
send receive on different frequencies based on the frequency plan of your
country. OTAA needs acks comming back from network server in strict time
windows ... single channel can't do that.