things4u/ESP-1ch-Gateway #108: How to decode sensor payload in TTNv3

wn1980, 2022-06-05:
Hi everybody.

I have successfully compiled the 1ch gateway as a sensor node (by setting `#define_GATEWAYNODE 1`) on the Heltec ESP32 LoRa v2. The gateway seems to send the uplink to TTN. However, I don't know how to decode the payload (such as battery value) in a human-readable format. Could anyone suggest a solution to this issue?

wn1980, 2022-06-05:
I found the Sensor format [document](https://things4u.github.io/Projects/SingleChannelGateway/DeveloperGuide/4%20_Lora_Message_Format/message_format.html). However, how to implement it in TTNv3 (`function decodeUplink(input)`) as described in [this page](https://www.thethingsindustries.com/docs/integrations/payload-formatters/javascript/uplink-decoder/).

Zupayruna1, 2022-08-20:
you need to check how the gateway are sending the data,
if the data is just pure bytes u can check this thread -> https://www.thethingsnetwork.org/forum/t/decrypting-messages-for-dummies/4894/29

I'm using this examples for encode and decode data