things4u/ESP-1ch-Gateway #110: ftoa precision

petergullberg, 2022-07-10:
Hi,
I struggled half-a-day to understand why the protocol didn't work, result is that I think I have found a defect.

_txRx.ino line 657
657: ftoa((double)freqs[gwayConfig.ch].upFreq / 1000000, cfreq, 6);		// XXX This can be done better

ftao generates "868.099975"....
This value is rejected TTN, and thus not forwarded...

Unclear if it's only ESP32, haven't spent more time on it...

\P

dennisheitmann, 2022-07-10:
Use this instead of old line 657: `__freqToCharMhz(freqs[gwayConfig.ch].upFreq, cfreq);`
with an additional function (place e.g. in **_utils.ino**): 
```
// ============== NUMBER FUNCTIONS ============================================

// ----------------------------------------------------------------------------
// Convert the given LoRa frequency, in Hz, to Mhz in a string
// It does not use float conversion, as float isnt accurate.
// Parameters:
//  freq is the freq to convert, from 100000000 to 999999999 Hz
//  *out is the strinf buffer, minimum 11 chars.
// ----------------------------------------------------------------------------
void __freqToCharMhz(uint32_t freq, char *out) {
  itoa(freq, out, 10);
  for (int i = 10; i > 2; i--){
    out[i] = out[i-1];
  }
  out[3] = '.';
}
```

petergullberg, 2022-07-11:
ok...

closed 2022-07-11 by petergullberg