things4u/ESP-1ch-Gateway #111: WifiManager Error

skyark21, 2022-07-24:
Hi i got this error when I try to compile it.
```[{
	"resource": "/home/skyark/Projects/Platformio/Heltec Gateway/lib/WiFiManager-development/WiFiManager.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no matching function for call to 'WiFiClass::onEvent(std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type)'",
	"startLineNumber": 2901,
	"startColumn": 68,
	"endLineNumber": 2901,
	"endColumn": 68
}]
```

I'm stuck and dont know what to do.

skyark21, 2022-07-24:
This is the complete output
```
Compiling .pio/build/Gateway_38/src/ESP-sc-gway.ino.cpp.o
In file included from lib/ESP8266_Oled_Driver_for_SSD1306_display/SSD1306Wire.h:31,
                 from lib/ESP8266_Oled_Driver_for_SSD1306_display/SSD1306.h:30,
                 from src/oLED.h:59,
                 from /home/skyark/Projects/Platformio/ESP-1ch-Gateway-master/src/ESP-sc-gway.ino:66:
lib/ESP8266_Oled_Driver_for_SSD1306_display/OLEDDisplay.h: In member function 'virtual bool OLEDDisplay::connect()':
lib/ESP8266_Oled_Driver_for_SSD1306_display/OLEDDisplay.h:253:29: warning: no return statement in function returning non-void [-Wreturn-type]
     virtual bool connect() {};
                             ^
Compiling .pio/build/Gateway_38/libba5/WiFiManager-development/WiFiManager.cpp.o
lib/WiFiManager-development/WiFiManager.cpp: In member function 'void WiFiManager::WiFiEvent(arduino_event_id_t, system_event_info_t)':
lib/WiFiManager-development/WiFiManager.cpp:3356:16: warning: comparison between 'enum arduino_event_id_t' and 'enum system_event_id_t' [-Wenum-compare]
   if (event == SYSTEM_EVENT_STA_DISCONNECTED)
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
lib/WiFiManager-development/WiFiManager.cpp:3372:21: warning: comparison between 'enum arduino_event_id_t' and 'enum system_event_id_t' [-Wenum-compare]
   else if (event == SYSTEM_EVENT_SCAN_DONE)
                     ^~~~~~~~~~~~~~~~~~~~~~
lib/WiFiManager-development/WiFiManager.cpp: In member function 'void WiFiManager::WiFi_autoReconnect()':
lib/WiFiManager-development/WiFiManager.cpp:3389:64: error: no matching function for call to 'WiFiClass::onEvent(std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type)'
   WiFi.onEvent(std::bind(&WiFiManager::WiFiEvent, this, _1, _2));
                                                                ^
In file included from /home/skyark/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:28,
                 from /home/skyark/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFi.h:32,
                 from lib/WiFiManager-development/WiFiManager.h:59,
                 from lib/WiFiManager-development/WiFiManager.cpp:13:
/home/skyark/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:159:21: note: candidate: 'wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventCb, arduino_event_id_t)'
     wifi_event_id_t onEvent(WiFiEventCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
                     ^~~~~~~
/home/skyark/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:159:21: note:   no known conversion for argument 1 from 'std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type' {aka 'std::_Bind<void (WiFiManager::*(WiFiManager*, std::_Placeholder<1>, std::_Placeholder<2>))(arduino_event_id_t, system_event_info_t)>'} to 'WiFiEventCb' {aka 'void (*)(arduino_event_id_t)'}
/home/skyark/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:160:21: note: candidate: 'wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventFuncCb, arduino_event_id_t)'
     wifi_event_id_t onEvent(WiFiEventFuncCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
                     ^~~~~~~
/home/skyark/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:160:21: note:   no known conversion for argument 1 from 'std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type' {aka 'std::_Bind<void (WiFiManager::*(WiFiManager*, std::_Placeholder<1>, std::_Placeholder<2>))(arduino_event_id_t, system_event_info_t)>'} to 'WiFiEventFuncCb' {aka 'std::function<void(arduino_event_id_t, arduino_event_info_t)>'}
/home/skyark/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:161:21: note: candidate: 'wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventSysCb, arduino_event_id_t)'
     wifi_event_id_t onEvent(WiFiEventSysCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
                     ^~~~~~~
/home/skyark/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:161:21: note:   no known conversion for argument 1 from 'std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type' {aka 'std::_Bind<void (WiFiManager::*(WiFiManager*, std::_Placeholder<1>, std::_Placeholder<2>))(arduino_event_id_t, system_event_info_t)>'} to 'WiFiEventSysCb' {aka 'void (*)(arduino_event_t*)'}
Compiling .pio/build/Gateway_38/FrameworkArduino/WString.cpp.o
Compiling .pio/build/Gateway_38/FrameworkArduino/base64.cpp.o
Compiling .pio/build/Gateway_38/FrameworkArduino/cbuf.cpp.o
Compiling .pio/build/Gateway_38/FrameworkArduino/esp32-hal-adc.c.o
Compiling .pio/build/Gateway_38/FrameworkArduino/esp32-hal-bt.c.o
Compiling .pio/build/Gateway_38/FrameworkArduino/esp32-hal-cpu.c.o
*** [.pio/build/Gateway_38/libba5/WiFiManager-development/WiFiManager.cpp.o] Error 1
```

Zupayruna1, 2022-08-20:
Maybe u need to check the libraries

mcenderdragon, 2022-10-12:
same problem. searching for ESP-WIfiManager like in the read me only made it worse (and there the heagder is called ESP_WifiManager.h and not just WifiManager.h)

nakagleo, 2022-11-15:
Hi just changed for https://github.com/tzapu/WiFiManager Lib, exluding the old WifiManager folder and including the one in the github and worked!!!