things4u/ESP-1ch-Gateway #118: Compile errors with PlatformIO

tk5ep, 2023-08-04:
HI, I'm trying to compile the ESP-1ch-Gateway with PlatformIO, but I get these errors : 
 `c:\users\admin\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\std_abs.h:78:3: note: candidate: 'constexpr long double std::abs(long double)'

   abs(long double __x)
   ^~~

c:\users\admin\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\std_abs.h:74:3: note: candidate: 'constexpr float std::abs(float)'
   abs(float __x)
   ^~~

c:\users\admin\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\std_abs.h:70:3: note: candidate: 'constexpr double std::abs(double)'
   abs(double __x)
   ^~~

c:\users\admin\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\std_abs.h:61:3: note: candidate: 'long long int std::abs(long long int)'
   abs(long long __x) { return __builtin_llabs (__x); }
   ^~~

c:\users\admin\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\std_abs.h:56:3: note: candidate: 'long int std::abs(long int)'
   abs(long __i) { return __builtin_labs(__i); }
   ^~~`

Does anyone has a solution.

Also, I would like to use a NodeMCU v3 board. Any luck to succeed ?

Thanks, Patrick.

MihaMarkic, 2023-08-05:
I've downloaded code as zip, compiled (on Windows) and it succeeds. Did you change anything? Using latest code?

tk5ep, 2023-08-05:
Hi Miha,

Thanks for responding.
I've downloaded the version which is on the main page and haven't changed anything at all for a first test.

I use  VSC 1.81.0 on Windows 10.

MihaMarkic, 2023-08-05:
By your error log it might be a problem with xtensa stuff.
I'm using these board settings, try using them and see if it compiles
```
platform = espressif32
board = ttgo-lora32-v1
framework = arduino
```

tk5ep, 2023-08-06:
I tried it, but with same bad results.
I do have these libraries. Maybe it is related.
No information about libraries are given and there are no lib_deps are declared.

|-- aes
|-- LoRaCode @ 1.0
|-- ArduinoJson @ 6.14.0
|-- ArduinoOTA @ 2.0.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- FS @ 2.0.0
|-- base64 @ 1.0.0
|-- SPI @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- Streaming @ 5.0.0
|-- Time @ 1.6
|-- TinyGPSPlus @ 1.0.2
|-- WebServer @ 2.0.0
|-- WiFi @ 2.0.0
|-- WifiManager @ 1.0.0
|-- ESP8266_SSD1306 @ 3.2.7

tk5ep, 2023-08-06:
Same results.

I tried to compile with the Arduino IDE. I also get a lot of errors despite all my corrections (libraries, wrong declarations)

I'm close to give up and I'm still looking for a working SCPF to make tests on the 433 MHz band.

MihaMarkic, 2023-08-06:
You have more libraries than me, odd, here are my (under lib directory)

aes
ArduinoJson
esp32-http-update
ESP8266_and_ESP32_Oled_Driver_for_SSD1306_display
ESP8266_Oled_Driver_for_SSD1306_display
gBase64
LoRaCode
Streaming
Time
TinyGPSPlus-1.0.2b
WiFiManager-development

and here is my only dependency declared in platformio.ini file

lib_deps =
  # RECOMMENDED
  # Accept new functionality in a backwards compatible manner and patches
  sandeepmistry/LoRa @ ^0.8.0

Try deleting your lib directory for starters and use my board configuration, I assume it will be recreated with just necessary libraries.

tk5ep, 2023-08-06:
Thanks for taking the time to try to help me.
Removing the libraries didn't help. I can't imagine how VSC can put the right libraries if there are no libs deps declared.
I'm not a PlatformIO specialist, so I'm a bit stucked.

With an empty lib folder, I compiled several times and resolved the library errors, one by one.
I ad to change some library names in the main sketch and load newer library versions until it worked.

Now it compiles further but I have this error : 
`*** [.pio\build\TK5EP\firmware.elf] Implicit dependency `C:\Users\Admin\.platformio\platforms\espressif32\builder\lib_deps' not found, needed by target `.pio\build\TK5EP\firmware.elf'.`

The libraries loaded are : 
Dependency Graph
|-- AES-128_V10
|-- ArduinoJson @ 6.14.0
|-- ArduinoOTA @ 2.0.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- FS @ 2.0.0
|-- base64 @ 1.0.0
|-- SPI @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- Streaming @ 5.0.0
|-- Time @ 1.6
|-- TinyGPSPlus @ 1.0.2
|-- WebServer @ 2.0.0
|-- WiFi @ 2.0.0
|-- WiFiManager @ 2.0.16-rc.2

Can you check if you have the same libs and versions ? (except for the WiFImanager I ihad to change, it didn't find a function in the original one)

Thanks,

tk5ep, 2023-08-06:
I managed to succeed to compile for a TTGO board.

But as soon as I change some settings in the 2 config files, it's going mad !
Dependency problems, some errors, etc...

Did you succeed to flash a TTGO board and make a gateway working ?

If yes, can you send me your confignode.h and configGway.h ?

But I think, I will give up.  :-(

tk5ep, 2023-08-06:
SUCCESS ! I managed to compile and make the gateway work, after a lot of testing and tweaking.
I had to scratch my head how to configure the whole stuff (WiFi settings, gateway ID, etc...)

It needed some time, but it finally works. I had to use the original config files and made only a very few adjustements.

The gateway reports to our local Chirpstark server.

But for the moment I can't decode my sensor board made with a demo sender software on an Arduino and sandeepmistry library.

Does someone have a sketch for Arduino that works to send LoRaWan packets ?

Thanks,

Bobkatze, 2023-08-24:
> 

Can you share your Projectfile? I have been trying to compile it for the T-Beam V1.1, and have fixed some errors, but cant progress from my current one.

tk5ep, 2023-08-25:
Hi Bob,

Glad to here that you managed to compile successfully.
Unfortunately, it looks like there is no support for this outdated package.

Yes, OLED works perfectly. I understand you have a T-beam I have a T3.
Did you correctly set the OLED type in configGway.h ?  #	define _OLED 1 as to be adjusted to the Oled type.

What band and frequency will you use ?

I can see my Gway receive packets, and forward them. But I can't see them on the Chirpstack server.
I need to be a bit more skilled in that.

Also deceoding SF12 doesn't work, SF7 works perfectly.

klaus-liebler, 2024-08-14:
My solution for the same compilation problem was in _txRx.ino:316:

`if (abs((int32_t)(freqs[gwayConfig.ch].dwnFreq - fff)) < 100000) {`

So, just add a typecast to `int32_t`.

HTH, regards Klaus