things4u/ESP-1ch-Gateway #119: Support for Heltec LoRa32 V3

Mouky66, 2023-08-26:
Hi,
is the Heltec LoRa32 V3 already supported or are you planing to implement the support?

Thank you
regards Chris

andrewbunac, 2023-11-03:
I am also waiting for this to happen, but for now, I am trying to edit the config file just to fit on a Heltec Lora V3. Hope somebody will make it happen :D

Samll, 2024-06-28:
Any update with this?
I'm trying but I get a WDG reset all the time:
st:0x8 (TG1WDT_SYS_RST),boot:0x29 (SPI_FAST_FLASH_BOOT)

klaus-liebler, 2024-08-14:
Same problem here...

```
UDP Connection successful
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
```

Btw, to solve compile issues I replaced "lib/WiFiManager-development" with the current version of this library and added some typecasting to _txRx.ino:316:
`if (abs((int32_t)(freqs[gwayConfig.ch].dwnFreq - fff)) < (int32_t)100000) {`

Thank you, regards Klaus